ethz-asl/ethzasl_xsens_driver #18: mtdevice.py: IndexError: string index out of range

kingfolk, 2015-11-06:
Hi, 
there is something wrong 
i run the following command and got `IndexError: string index out of range`,

```
./mtdevice.py --echo --device=/dev/ttyUSB0 --output-mode=o --output-settings=q
```

Full stack traceback:

```
Traceback (most recent call last):
  File "./mtdevice.py", line 1124, in <module>
    main()
  File "./mtdevice.py", line 1017, in main
    print mt.read_measurement(mode, settings)
  File "./mtdevice.py", line 400, in read_measurement
    return self.parse_MTData2(data)
  File "./mtdevice.py", line 591, in parse_MTData2
    data = data[3+size]
IndexError: string index out of range
```
I don't know what data is storing. `print data` returns something miss coded string, like

```
 0
  >�A�}&Cc�
```
and the data is different every time.

also `roslaunch xsens_driver xsens_driver.launch` return the same thing.

Dev environment is Ubuntu 14.04 kernel 3.19.3 ros indigo

I can run mtmanager to get the roll/pitch/yaw and orientation data.

the following is `lsusb -v -d id` result. 
```
Bus 001 Device 008: ID 2639:0003
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x2639
  idProduct          0x0003
  bcdDevice            0.00
  iManufacturer           1 Xsens
  iProduct                2 MTi-30 AHRS
  iSerial                 3 0360048E
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           48
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

fcolas, 2015-11-06:
I believe you're running an old version and that this issue was fixed by #13; can you try to update to the current master?

kingfolk, 2015-11-06:
cool! great work!
seems okay after i build from source!

fcolas, 2015-11-06:
Cool, thanks.

closed 2015-11-06 by fcolas

kingfolk, 2015-11-06:
ok. some other thing pop up.
```
rosrun xsens_driver mtdevice.py --configure --output-mode=coapvs --output-settings=tqMAG
```
returns `Configuring mode and settings MT error: timeout waiting for message.`

fcolas, 2015-11-06:
Well, there are sometimes some issues that I'd need to investigate: it seems that the IMU doesn't always give the expected acknowledgment but might still have received the message.

You can try to check that the device actually changed its configuration as you wanted:

    ./mt_device.py -i

Otherwise you can try to change the `verbose` flag line 10 to see what is going on.

kingfolk, 2015-11-06:
Thanks!

The imu i am using is a MTi 30. And i did not configure it but i don't know if somebody else has.
could you check the verbose input for me? great thanks!

```
Device: /dev/ttyUSB0 at 115200 Bd:
General configuration: {'output-settings': 1, 'device ID': 56624270, 'length': 0, 'skipfactor': 0, 'number of devices': 1, 'time': '\x00\x00\x00\x00\x00\x00\x00\x00', 'date': '\x00\x00\x00\x00\x00\x00\x00\x00', 'output-mode': 6, 'Master device ID': 56624270, 'period': 1152}
Available scenarios: [(39, 2, 'general'), (40, 2, 'high_mag_dep'), (41, 2, 'dynamic'), (42, 2, 'low_mag_dep'), (43, 2, 'vru_general')]
Current scenario: general (id: 39)
coman@coman-head:~$ rosrun xsens_driver mtdevice.py -c -p 576 -m co -s nqAGM
Configuring mode and settings MT error: timeout waiting for message.
coman@coman-head:~$ rosrun xsens_driver mtdevice.py -iMT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 04 8E 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 04 8E 00 00 00 06 00 00 00 01 00 27 01 04 01 49 02 00]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
Device: /dev/ttyUSB0 at 115200 Bd:
General configuration: MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 04 8E 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 04 8E 00 00 00 06 00 00 00 01 00 27 01 04 01 49 02 00]
{'output-settings': 1, 'device ID': 56624270, 'length': 0, 'skipfactor': 0, 'number of devices': 1, 'time': '\x00\x00\x00\x00\x00\x00\x00\x00', 'date': '\x00\x00\x00\x00\x00\x00\x00\x00', 'output-mode': 6, 'Master device ID': 56624270, 'period': 1152}
Available scenarios: MT: Write message id 0x62 (ReqAvailableScenarios) with 0 data bytes: []
MT: Got message id 0x63 (AvailableScenarios) with 110 data bytes: [27 02 67 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20 20 20 20 20 28 02 68 69 67 68 5F 6D 61 67 5F 64 65 70 20 20 20 20 20 20 20 20 29 02 64 79 6E 61 6D 69 63 20 20 20 20 20 20 20 20 20 20 20 20 20 2A 02 6C 6F 77 5F 6D 61 67 5F 64 65 70 20 20 20 20 20 20 20 20 20 2B 02 76 72 75 5F 67 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20]
[(39, 2, 'general'), (40, 2, 'high_mag_dep'), (41, 2, 'dynamic'), (42, 2, 'low_mag_dep'), (43, 2, 'vru_general')]
MT: Write message id 0x64 (SetCurrentScenario) with 0 data bytes: []
MT: Got message id 0x65 (SetCurrentScenarioAck) with 2 data bytes: [00 27]
Current scenario: general (id: 39)
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
coman@coman-head:~$
```
it seems MT always write empty message.

fcolas, 2015-11-09:
Ok, thanks for trying out the verbose flag, but it'd be even better if you'd do it on the command that gives the error message (setting the configuration). ;)

As for 0 bytes messages, that's expected so nothing to worry about. There is one thing that is unexpected: you ask to not have timestamp but in the settings you still have a sample count. But the output is properly set to both calibrated and orientation data.

But I'd like to know already whether you're able to get the data you want or not?

kingfolk, 2015-11-09:
The output is for `rosrun xsens_driver mtdevice.py -i`. So it is okay to have very verbose information right?

I

fcolas, 2015-11-09:
Yes, I've seen that the verbose output was for the inspection mode `-i` but your problem is with the configuration `-c` just before.

kingfolk, 2015-11-09:
ok. Thanks!

is there anywhere i can debug about what is wrong with my setup?

fcolas, 2015-11-09:
Isn't it what we're trying to do here? ^ ^
Can you put `verbose = True` line 10 then run `rosrun xsens_driver mtdevice.py -c -p 576 -m co -s nqAGM` and paste the (verbose) output here?

kingfolk, 2015-11-09:
sure! here's output. i mean if there is some place in code i can set a breakpoint.
```
coman@coman-head:~$ rosrun xsens_driver mtdevice.py -c -p 576 -m co -s nqAGM
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x36 (MTData2) with 15 data bytes: [20 30 0C BD A2 80 E0 BF 08 43 BC C3 2E 4B 6E]
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 04 8E 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 04 8E 00 00 00 06 00 00 00 01 00 27 01 04 01 49 02 00]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
Configuring mode and settings MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0xD0 (SetOutputMode) with 2 data bytes: [00 06]
MT error: timeout waiting for message.
```

fcolas, 2015-11-09:
Well, that gives us some information already: we're normally able to exchange messages with the IMU but when changing the configuration we don't get the expected acknowledgement.
Now there are two questions: did the IMU take the change of configuration into account and, if so, why don't we get an acknowledgement.

To answer the first question, can you try to run (to see if the change happens even when we don't get the ACK):
`rosrun xsens_driver mtdevice.py -c -p 576 -m c -s nqAGM`
`rosrun xsens_driver mtdevice.py -i`

Can you also change the number of tries in line 114 to something like 50, to see if it solves the issue?

kingfolk, 2015-11-09:
Thanks.. i am also altering that part of code but to the wrong direction...
```
coman@coman-head:~$ rosrun xsens_driver mtdevice.py -c -p 576 -m co -s nqAGM
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 04 8E 02 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 04 8E 00 00 00 06 00 00 00 00 00 27 01 04 01 49 02 00]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
Configuring mode and settings MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0xD0 (SetOutputMode) with 2 data bytes: [00 06]
MT: Got message id 0xD1 (SetOutputModeAck) with 0 data bytes: []
MT: Write message id 0xD2 (SetOutputSettings) with 4 data bytes: [00 00 00 00]
MT: Got message id 0xD3 (SetOutputSettingsAck) with 0 data bytes: []
MT: Write message id 0x04 (SetPeriod) with 2 data bytes: [02 40]
MT: Got message id 0x05 (SetPeriodAck) with 0 data bytes: []
MT: Write message id 0xD0 (SetOutputMode) with 0 data bytes: []
MT: Got message id 0xD1 (SetOutputModeAck) with 2 data bytes: [00 06]
MT: Write message id 0xD2 (SetOutputSettings) with 0 data bytes: []
MT: Got message id 0xD3 (SetOutputSettingsAck) with 4 data bytes: [00 00 00 00]
MT: Write message id 0x0A (ReqDataLength) with 0 data bytes: []
MT error 0x04: Invalid message.MT: Got message id 0x42 (Error) with 5 data bytes: [04 00 00 00 00]
MT error: timeout waiting for message.
coman@coman-head:~$ rosrun xsens_driver mtdevice.py -i
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 04 8E 02 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 04 8E 00 00 00 06 00 00 00 00 00 27 01 04 01 49 02 00]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
Device: /dev/ttyUSB0 at 115200 Bd:
General configuration: MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 04 8E 02 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 04 8E 00 00 00 06 00 00 00 00 00 27 01 04 01 49 02 00]
{'output-settings': 0, 'device ID': 56624270, 'length': 0, 'skipfactor': 0, 'number of devices': 1, 'time': '\x00\x00\x00\x00\x00\x00\x00\x00', 'date': '\x00\x00\x00\x00\x00\x00\x00\x00', 'output-mode': 6, 'Master device ID': 56624270, 'period': 576}
Available scenarios: MT: Write message id 0x62 (ReqAvailableScenarios) with 0 data bytes: []
MT: Got message id 0x63 (AvailableScenarios) with 110 data bytes: [27 02 67 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20 20 20 20 20 28 02 68 69 67 68 5F 6D 61 67 5F 64 65 70 20 20 20 20 20 20 20 20 29 02 64 79 6E 61 6D 69 63 20 20 20 20 20 20 20 20 20 20 20 20 20 2A 02 6C 6F 77 5F 6D 61 67 5F 64 65 70 20 20 20 20 20 20 20 20 20 2B 02 76 72 75 5F 67 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20]
[(39, 2, 'general'), (40, 2, 'high_mag_dep'), (41, 2, 'dynamic'), (42, 2, 'low_mag_dep'), (43, 2, 'vru_general')]
MT: Write message id 0x64 (SetCurrentScenario) with 0 data bytes: []
MT: Got message id 0x65 (SetCurrentScenarioAck) with 2 data bytes: [00 27]
Current scenario: general (id: 39)
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
coman@coman-head:~$
```
seem the error is invalid message.

fcolas, 2015-11-09:
Oh, that's interesting.
First, we get further away, which is rather good news. Second, now the `output-settings` are properly set. Third, it doesn't like to be requested the size of the message (and that's unexpected).

But that part is actually not important at this stage. Are you able to `rosrun xsens_driver mtdevice.py -e`?
If so, then we may have an unexpected error but without practical consequences for you.
Is the node running properly (you can revert `verbose` to `False`)?

kingfolk, 2015-11-09:
Yes! i can run that command!
i get so many `{'Orientation Data': {'Yaw': 149.8646240234375, 'Roll': 0.04704917594790459, 'Pitch': -0.365906298160553}}  `. And if i rotate imu, the output changes.

fcolas, 2015-11-09:
Ok, that seems to work, then. I'll try see where the "invalid message" error comes from but you can safely ignore it in the configuration.

kingfolk, 2015-11-09:
cool, thanks! so is it now possible to get the xsens_dirver running and bypass this invalid message?

fcolas, 2015-11-10:
Well, as you can see, this message happens at the configuration command. But as the IMU stores its configuration, you don't need run this command every time. The ROS driver doesn't have this issue.

kingfolk, 2015-11-10:
Thanks. But sorry that i don't understand your words. could you give me more instructions?
`-e` just let me print out the imu data right? it did not change any configuration. So the old configuration still is there and xsens_dirver read that configuration. In fact, when i try `rostopic echo /imu/data`, there is no output.

fcolas, 2015-11-10:
Well, I don't like saying that, but I feel that if you read carefully the [documentation](http://wiki.ros.org/xsens_driver), you should be able to understand what I wrote, and especially the difference between the node and the configuration tool. It's probably useless to repeat what's already written so check it again, use the node, and ask a specific question if something is unclear.

kingfolk, 2015-11-10:
Thanks a lot.
The problem is the node not stream data to ros topic. `roslaunch xsens_driver xsens_driver.launch` and then `rostopic echo /imu/data`.

That is my mtdevice configuration.
```
root@coman-head:~# rosrun xsens_driver mtdevice.py -i
Device: /dev/ttyUSB0 at 115200 Bd:
General configuration: {'output-settings': 1, 'device ID': 56624270, 'length': 0, 'skipfactor': 0, 'number of devices': 1, 'time': '\x00\x00\x00\x00\x00\x00\x00\x00', 'date': '\x00\x00\x00\x00\x00\x00\x00\x00', 'output-mode': 2110, 'Master device ID': 56624270, 'period': 576}
Available scenarios: [(39, 2, 'general'), (40, 2, 'high_mag_dep'), (41, 2, 'dynamic'), (42, 2, 'low_mag_dep'), (43, 2, 'vru_general')]
Current scenario: general (id: 39)
```

fcolas, 2015-11-10:
Oh, weird. What happens if you launch the new node (`rosrun xsens_driver mtnode_new.py`), rather than the old one: do you have output on `/imu/data`?
With the new configuration, does `rosrun xsens_driver mtdevice.py -e` still provide info?

kingfolk, 2015-11-10:
`rosrun xsens_driver mtnode_new.py` works! i got the `/imu/data` finally! 

thanks man! And sorry that i asked some vague questions.

fcolas, 2015-11-10:
No problem, I'm glad that it works. I'll try to free some time soon to clean a few things and deprecate `mtnode.py`.