ethz-asl/ethzasl_xsens_driver #20: Support for reading extended Motion Tracker Specification data (eMTS) and logging to file

herrfrei, 2016-01-07:
Hi, to create a inpuit file for the magnetic field mapper I needed to change mtdef.py and mtdevice.py in the following manner:
mtdef.py: 
  - ReqEmts = 0x90 and Emts = 0x91 added to class MID
mtdevice.py: changes to MTDevice class
  - corrected CRC computation error when extended length is used (needed for eMTS data) in read_msg()
  - added log file writing in read_msg()
  - added a member logfile and a function set_logfile() to set it
  - added function ReqEmts() to request the eMTS data
get_calib_data.py: new file containing the calibration code. call it with -d DEVICE_NAME to create a log file (supports also setting baudrate and verbose mode), press CTRL-C to stop recording

It would be fine if these changes could be integrated into the driver code. I append the corrected files.

[pyXsensCalibrator.zip](https://github.com/ethz-asl/ethzasl_xsens_driver/files/80993/pyXsensCalibrator.zip)

anlif, 2016-01-07:
It's much easier for the repository owners to add the changes if you create a pull request. See https://help.github.com/articles/using-pull-requests/

fcolas, 2016-01-07:
@anlif is right: please submit a pull request.

fcolas, 2016-08-02:
See #21

closed 2016-08-02 by fcolas