ethz-asl/ethzasl_xsens_driver #23: GPS data on /fix and /fix_extended topics are not advertised with MTi-G-700

Abdallah-Sobehy, 2016-05-10:
I had the MT `error: timeout waiting for message.` issue with the mtdevice.py node as in the [issue](https://github.com/ethz-asl/ethzasl_xsens_driver/issues/18). Following the discussion, I checked that the configuration occurs anyways by changing the period with `-p` and seeing the effect with the `-i` command. So, ignoring the error message in the configuration step and running mtnode_new I am able to see the imu/data but nothing is returned with /fix , /fix_extended and /temperature. However, I am only interested in the GPS data.

Here is the configuration command I run :

> rosrun xsens_driver mtdevice.py --configure --output-mode=tcoapvsgr --output-settings=tqMAG
 
Corresponding output :

> MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [07 70 01 B1 02 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 07 70 01 B1 00 00 58 3F 00 00 00 01 04 01 01 03 01 37 02 02]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
Configuring mode and settings MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0xD0 (SetOutputMode) with 2 data bytes: [58 3F]
MT: Got message id 0xD1 (SetOutputModeAck) with 0 data bytes: []
MT: Write message id 0xD2 (SetOutputSettings) with 4 data bytes: [00 00 00 01]
MT: Got message id 0xD3 (SetOutputSettingsAck) with 0 data bytes: []
MT: Write message id 0xD0 (SetOutputMode) with 0 data bytes: []
MT: Got message id 0xD1 (SetOutputModeAck) with 2 data bytes: [58 3F]
MT: Write message id 0xD2 (SetOutputSettings) with 0 data bytes: []
MT: Got message id 0xD3 (SetOutputSettingsAck) with 4 data bytes: [00 00 00 01]
MT: Write message id 0x0A (ReqDataLength) with 0 data bytes: []
MT error 0x04: Invalid message.MT: Got message id 0x42 (Error) with 5 data bytes: [04 00 00 00 00]
MT error: timeout waiting for message.

Also here is the inspection command I run :

> rosrun xsens_driver mtdevice.py -i

And the corresponding output: 

> MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [07 70 01 B1 02 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 07 70 01 B1 00 00 58 3F 00 00 00 01 04 01 01 03 01 37 02 02]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
Device: /dev/ttyUSB0 at 115200 Bd:
General configuration: MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [07 70 01 B1 02 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 07 70 01 B1 00 00 58 3F 00 00 00 01 04 01 01 03 01 37 02 02]
{'output-settings': 1, 'device ID': 124780977, 'length': 0, 'skipfactor': 0, 'number of devices': 1, 'time': '\x00\x00\x00\x00\x00\x00\x00\x00', 'date': '\x00\x00\x00\x00\x00\x00\x00\x00', 'output-mode': 22591, 'Master device ID': 124780977, 'period': 576}
Available scenarios: MT: Write message id 0x62 (ReqAvailableScenarios) with 0 data bytes: []
MT: Got message id 0x63 (AvailableScenarios) with 110 data bytes: [01 04 47 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20 20 20 20 20 02 04 47 65 6E 65 72 61 6C 4E 6F 42 61 72 6F 20 20 20 20 20 20 20 03 04 47 65 6E 65 72 61 6C 4D 61 67 20 20 20 20 20 20 20 20 20 20 04 04 41 75 74 6F 6D 6F 74 69 76 65 20 20 20 20 20 20 20 20 20 20 05 04 41 75 74 6F 55 72 62 61 6E 43 61 6E 79 6F 6E 20 20 20 20 20]
[(1, 4, 'General'), (2, 4, 'GeneralNoBaro'), (3, 4, 'GeneralMag'), (4, 4, 'Automotive'), (5, 4, 'AutoUrbanCanyon')]
MT: Write message id 0x64 (SetCurrentScenario) with 0 data bytes: []
MT: Got message id 0x65 (SetCurrentScenarioAck) with 2 data bytes: [04 01]
Current scenario:  (id: 1025)
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []

Thanks a lot for the effort you invest in aiding others, and hope you can help me getting the GPS data to be read.

fcolas, 2016-05-10:
@Abdallah-Sobehy can you tell which model of XSens IMU do you have?
Also, I see that you're requesting everything however the documentation states that raw inertial measurement is only compatible with raw GPS. If you're only interested in GPS as you say, what happens when you ask only for it?

Abdallah-Sobehy, 2016-05-11:
I am not quite sure, what do you mean by which model, I mentioned in the issue title that I am using the MTi-G-700, isn't this what you are asking for ?
I meant by only interested in GPS that I do not need the temperature. I tried the again the output mode without `r` option as follows : 

> rosrun xsens_driver mtdevice.py --configure --output-mode=coapvsg --output-settings=tqMAG

and still nothing is advertised on for or fix_extended
and with only GPS data as follows : 

> rosrun xsens_driver mtdevice.py --configure --output-mode=pg --output-settings=tqMAG

What is noticble here is that I could still see data from `imu/data` topic. Shouldn't this be stopped in the configuration step ?

fcolas, 2016-05-11:
Ok, sorry, I had missed the title.
I guess that you confirmed that the configuration with only GPS (let's say `--output-mode=p` for the sake of brevity) was correctly set in the device as you did in your first message.
The question is then whether the device is actually outputting the expected information or not. Can you show the debug output of:
> rosrun xsens_driver mtdevice.py

Abdallah-Sobehy, 2016-05-11:
With verbose set to true, I get the following output when running `rosrun xsens_driver mtdevice.py`

> MT: Got message id 0x36 (MTData2) with 85 data bytes: [10 60 04 03 29 E7 AB 80 20 0C BC 3E 6C 4B B9 E9 BD 1C 3B 18 70 5D 40 20 0C BD 53 35 AF BE 48 25 66 41 1D 10 D0 C0 20 0C 3C E2 EF 1E BE FD 0D E3 BF 6E 47 3A 30 10 04 00 01 6F D1 E0 20 04 01 80 00 87 20 10 10 3F 7F CC 6D BD 0B FA F6 3B 1D F7 EB 3C A3 BF C7]
{'Acceleration': {'accY': -0.1954551637172699, 'accX': -0.05156486853957176, 'accZ': 9.816604614257812}, 'Status': {'StatusWord': 25165959}, 'Timestamp': {'SampleTimeFine': 53077931}, 'Magnetic': {'magZ': -0.9307743310928345, 'magY': -0.49424657225608826, 'magX': 0.027701910585165024}, 'Angular Velocity': {'gyrZ': 0.0023260333109647036, 'gyrX': -0.011622498743236065, 'gyrY': -0.0004458212060853839}, 'Pressure': {'Pressure': 94161}, 'Orientation Data': {'Q1': -0.034174881875514984, 'Q0': 0.9992130398750305, 'Q3': 0.019988907501101494, 'Q2': 0.00241040694527328}}

fcolas, 2016-05-11:
Weird. Can you also put the beginning of the output?
And the debug output of the configuration command as well?

Abdallah-Sobehy, 2016-05-11:
> MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [07 70 01 B1 02 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 07 70 01 B1 00 00 00 10 00 00 00 01 04 01 01 03 01 37 02 02]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
MT: Got message id 0x36 (MTData2) with 66 data bytes: [10 60 04 03 B2 92 31 80 20 0C BA 6C 13 0C 37 55 BA 19 39 33 22 C8 40 20 0C BA 68 88 94 BC 91 51 76 3F 3C 68 28 C0 20 0C 3C D5 33 FA BE F7 DF 91 BF 6E D2 42 30 10 04 00 01 6F C1 E0 20 04 01 80 00 87]
{'Acceleration': {'accY': -0.017739038914442062, 'accX': -0.0008870449382811785, 'accZ': 0.735964298248291}, 'Status': {'StatusWord': 25165959}, 'Timestamp': {'SampleTimeFine': 62034481}, 'Magnetic': {'magZ': -0.9328957796096802, 'magY': -0.48412755131721497, 'magX': 0.026025760918855667}, 'Angular Velocity': {'gyrZ': 0.00017083727288991213, 'gyrX': -0.0009005523752421141, 'gyrY': 1.2739118574245367e-05}, 'Pressure': {'Pressure': 94145}}

fcolas, 2016-05-11:
Well, thanks for the log. What I can see is that the output mode is correctly set but the data output doesn't correspond to it. According to the documentation I have, that shouldn't happen: `SetOutputMode` (which was the way to configure mk III devices) is supposed to be recognized by newer devices (mk. IV).
If you have access to to a Windows computer with the vendor software, you can try to set it to legacy mode which might solve the problem.

Abdallah-Sobehy, 2016-05-12:
Okay, thanks for your suggestion, I will get access to a windows computer as soon as I can and update you with the outcome in legacy mode. So, if it does not work would you think it is a hardware problem with the GPS sensor ?

fcolas, 2016-05-20:
Ok, I was able to confirm that new models with new firmwares only abide by the configuration of `SetOutputMode` in legacy mode.
I'll try to update the node and driver to better support the mark IV devices in the near future.

fcolas, 2016-08-02:
This should be solved by #26

closed 2016-08-02 by fcolas