ethz-asl/ethzasl_xsens_driver #24: Issue with  MTi 1-series dev kit drivers after firmware update

gnarang1024, 2016-05-11:
Hello, I've been using your these drivers with a [MTi 1-series Development Kit](https://www.xsens.com/download/pdf/documentation/mti-1/mti-1-series_dk_user_manual.pdf) (with the MTi-3-8A7G6 sensor) for some time without any problem. 

I recently updated the firmware from FW 1.0.6 version to the lates, FW 1.0.12 version and now, when I launch the xsens_driver.launch file I'm seeing the following error:

> 
>  SUMMARY
> 
> ========
> 
> PARAMETERS
>  * /rosdistro: indigo
>  * /rosversion: 1.11.19
>  * /xsens_driver/frame_id: imu
>  * /xsens_driver/frame_local: ENU
>  * /xsens_driver/frame_local_imu: ENU
> 
> NODES
>   /
>     xsens_driver (xsens_driver/mtnode_new.py)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> core service [/rosout] found
> process[xsens_driver-1]: started with pid [25162]
> [WARN] [WallTime: 1462986141.530980] Cannot find value for parameter: ~device, assigning default: auto
> [WARN] [WallTime: 1462986141.531970] Cannot find value for parameter: ~baudrate, assigning default: 0
> [INFO] [WallTime: 1462986141.621873] Detected MT device on port /dev/ttyUSB0 @ 115200 bps
> [INFO] [WallTime: 1462986141.622076] MT node interface: /dev/ttyUSB0 at 115200 bd.
> [INFO] [WallTime: 1462986141.663337] Found parameter: ~frame_id, value: imu
> Traceback (most recent call last):
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 495, in <module>
>     main()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 491, in main
>     driver.spin()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 111, in spin
>     self.spin_once()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 448, in spin_once
>     data = self.mt.read_measurement()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 390, in read_measurement
>     mid, data = self.read_msg()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 120, in read_msg
>     if ord(waitfor())<>0xFA:
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 118, in waitfor
>     raise MTException("timeout waiting for message.")
> mtdef.MTException: MT error: timeout waiting for message.

I'm running these drivers in ROS Indigo and everything was running perfectly with. the previous firmware!

I also tried testing the IMU in the MT software suite V4.5.4 provided by xsens [MT_Software_Suite_Linux_4.5.4](https://www.xsens.com/mt-software-suite/) with the latest firmware(FW 1.0.12). Even though I was able to visualize the data from the IMU, I was getting the following out on my terminal window:

> 
>  ./mtmanager 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(351) setLogLevel Log level switched to [ERROR] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(365) setDebugLevel Debugger output log level switched to [FATAL] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(207) writeFileHeader Journaller logging to /home/gautam/.local/share/Xsens/mtmanager/xda.log for xda
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(208) writeFileHeader Current log level is [ERROR] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(212) writeFileHeader Journaller is not using threaded logging
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(365) setDebugLevel Debugger output log level switched to [FATAL] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(207) writeFileHeader Journaller logging to /home/gautam/.local/share/Xsens/mtmanager/mtmanager.log for MT Manager
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(208) writeFileHeader Current log level is [ERROR] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(212) writeFileHeader Journaller is not using threaded logging
> 1462986441.328 <3EBE4700> [WRITE] createMasterDevice id: 03880223 firmware version: 1.0.12 build 106 rev 52752
>  

Please let me know if you have any suggestion for me. I'll be happy to provide you with any other information that might be helpful. Thanks in advance!

jgonzal13, 2016-05-18:
hi @gnarang1024, 

have you made in progress on this issue? I also recently started using the same sensor (product code: MTI-3-8A7G6), but I'm also having trouble with the xsens_driver package with this specific sensor.

I haven't tried updating the firmware since I got it, but if I try to run the xsens_driver launch file (with the 'baudrate' and 'device' parameters set in the launch file), the node quickly dies because of some configuration error. I get the output  below.

have you encountered this error? if so, any suggestions on how to fix it? you mentioned you modified the driver?

thanks!
____________________________________________________________________
SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.19
 * /xsens_driver/baudrate: 115200
 * /xsens_driver/device: /dev/ttyUSB0
 * /xsens_driver/frame_id: /imu
 * /xsens_driver/frame_local: ENU
 * /xsens_driver/frame_local_imu: ENU

NODES
  /
    xsens_driver (xsens_driver/mtnode.py)

auto-starting new master
process[master]: started with pid [18836]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to abd9a790-1d42-11e6-ac5e-0024e848044e
process[rosout-1]: started with pid [18849]
started core service [/rosout]
process[xsens_driver-2]: started with pid [18852]
[INFO] [WallTime: 1463608287.940242] Found parameter: ~device, value: /dev/ttyUSB0
[INFO] [WallTime: 1463608287.941118] Found parameter: ~baudrate, value: 115200
[WARN] [WallTime: 1463608287.941915] Cannot find value for parameter: ~timeout, assigning default: 0.002
[INFO] [WallTime: 1463608287.942067] MT node interface: /dev/ttyUSB0 at 115200 bd.
Traceback (most recent call last):
  File "/home/jgon14/workspace/src/ethzasl_xsens_driver/nodes/mtnode.py", line 348, in <module>
    main()
  File "/home/jgon14/workspace/src/ethzasl_xsens_driver/nodes/mtnode.py", line 343, in main
    driver = XSensDriver()
  File "/home/jgon14/workspace/src/ethzasl_xsens_driver/nodes/mtnode.py", line 60, in __init__
    self.mt = mtdevice.MTDevice(device, baudrate, timeout)
  File "/home/jgon14/workspace/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 30, in __init__
    self.auto_config()
  File "/home/jgon14/workspace/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 375, in auto_config
    self.ReqConfiguration()
  File "/home/jgon14/workspace/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 280, in ReqConfiguration
    self.header = '\xFA\xFF\x32'+chr(length)
ValueError: chr() arg not in range(256)

gnarang1024, 2016-06-14:
So, the problem was that the timeout parameter was not enough. The default is set to (1/samplerate + 6ms). Increasing it to 10ms in the mtdef source file -->'additionalTimeOutOffset' solved the probelm.
Please see : https://github.com/xsens/xsens_mti_ros_node/issues/8 for details.
-Gautam

fcolas, 2016-08-02:
Many changes have occurred in the past weeks. Can you check again with the master branch and report if you still have the issue?

fcolas, 2017-03-09:
If I don't get any news on this, I'm probably going to close this issue as it doesn't seem to be relevant any more.

closed 2017-04-10 by fcolas