ethz-asl/ethzasl_xsens_driver #31: Run command SetNoRotation

athtest800, 2016-12-04:
Hello!
How to run command SetNoRotation at startup so to calibrate the MTi-3?
It is in mtdevice.py
Thank you

fcolas, 2016-12-05:
Well, so far, this method isn't used anywhere. You can always use `mtdevice` as a module in an interactive Python session and manually call this function but I believe it's not what you want.

If you'd like to have this run each time you launch the ROS node, I guess the cleanest would be to have a new node parameter `~no_rotation_duration` defaulting to 0 and invoke the SetNoRotation procedure if it's not 0.
Would that suit you? Can you try it out and propose a pull request?

fcolas, 2016-12-07:
Ok, I did it. Can you try the code on the [no_rotation_on_startup](https://github.com/ethz-asl/ethzasl_xsens_driver/tree/no_rotation_on_startup) branch and tell me if it solves your issue?

fcolas, 2017-03-09:
Any news on that front? Where you able to test the branch?

athtest800, 2017-03-19:
It works only if I set seconds as String
`self.mt.SetNoRotation('10')`
It does not work if
`self.mt.SetNoRotation(10)`

fcolas, 2017-03-22:
Are you sure of that? I cannot test it here but from my code, `self.mt.SetNoRotation('10')` should give an exception from `struct.pack` such as `cannot convert argument to integer`.
When you say it does not work with `SetNoRotation(10)` what is the error?

athtest800, 2017-04-11:
Hi! Sorry, it was my fault. It runs OK!!
Thank you again!!

closed 2017-04-11 by athtest800