ethz-asl/ethzasl_xsens_driver #32: Support for MTw2 devices

metzge-ka, 2017-01-23:
In my current project, I have to use a Raspberry Pi 3 (RPi) with the Raspbian OS (version: jessie). Also some Xsens MTw2 sensors (up to 6) have to be connected to the RPi and the goal will be to record all sensor data to a SD card.
At first I tried to use predefined software from Xsens, such as the MT Manager, the Xsens Device API and accompanying example code. But, it turned out that it is not possible to use this on the RPi, because of the ARM architecture (but the sensors have full Linux support according to the Xsens homepage!).
That's why I started to use ROS with some Node xsens_mti_ros_node-master, which is supposed to work. The node provided by Xsens does not work with the MTw2 sensors, though!
Then I was happy to find the original author/maintainer of the ROS Node.

Starting the node and calling the config of the MTw2 sensor as follows, seems to work - more or less - with the MTw2 sensor, but the inspection stops there and we don't get the full configuration:
$ roslaunch src/ethzasl_xsens_driver-master/launch/xsens_driver.launch
and then:
$ ./mtdevice.py -b 0 -v -i
The output from the MTi compared to the MTw suggests that something is wrong (see attatchment).
[MTi MTw.docx](https://github.com/ethz-asl/ethzasl_xsens_driver/files/722881/MTi.MTw.docx)

Update: Changing the .launch file to:
[xsens_driver_2.zip](https://github.com/ethz-asl/ethzasl_xsens_driver/files/722899/xsens_driver_2.zip)
works with 2 MTw2 sensors.

Great work with the driver/wrapper. Many thanks!

fcolas, 2017-01-23:
Dear Kai,

Thanks for the report. I'd like to investigate what's wrong with the inspection.
Could you try to comment out the line [1519](https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtdevice.py#L1519) of `mtdevice.py` to see whether this request is the only one posing problem?
It'd be nice if you could comment out each blocking request and tell me which are working and which are not.

(BTW, thanks for the colors, but I strongly prefer a simple text file and, when it's not too long, it's easier if you just paste it here.)

metzge-ka, 2017-01-24:
Dear Francis,

Here are the outputs your requested:

1) commenting out line 1519:
pi@raspberrypi:~/catkin_ws/src/ethzasl_xsens_driver-master/nodes $ ./mtdevice.py -i
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x00B416B7
  product code: 'MTW2-3A7G6'
  firmware revision: (4, 0, 2)
  baudrate: 2
Traceback (most recent call last):
  File "./mtdevice.py", line 1704, in <module>
    main()
  File "./mtdevice.py", line 1430, in main
    inspect(mt, device, baudrate)
  File "./mtdevice.py", line 1520, in inspect
    try_message("option flags:", mt.GetOptionFlags, hex_fmt(8))
AttributeError: 'MTDevice' object has no attribute 'GetOptionFlags'

2) commenting out line 1519 & 1520:
pi@raspberrypi:~/catkin_ws/src/ethzasl_xsens_driver-master/nodes $ ./mtdevice.py -i
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x00B416B7
  product code: 'MTW2-3A7G6'
  firmware revision: (4, 0, 2)
  baudrate: 2
  location ID:  0x0000
  transmit delay:  MTException: Timeout: waiting for message

3) commenting out line 1519 & 1429:1430:
no output

4) ./mtdevice.py -e works fine:
{'Acceleration': {'Delta v.z': 0.09768372774124146, 'Delta v.y': -0.0005879998207092285, 'Delta v.x': 0.0014594793319702148, 'frame': 'ENU', 'accY': -0.05904475599527359, 'accX': 0.14560505747795105, 'accZ': 9.768375396728516}, 'Timestamp': {'PacketCounter': 208, 'SampleTimeCoarse': 2, 'SampleTimeFine': 21037}, 'Magnetic': {'magZ': -0.8065967559814453, 'frame': 'ENU', 'magX': -0.0055239200592041016, 'magY': 0.5240352153778076}, 'Angular Velocity': {'Delta q1': -1.2155623153375927e-05, 'Delta q0': 1.0000001192092896, 'Delta q3': 2.5044195353984833e-05, 'Delta q2': 1.7399899661540985e-05, 'frame': 'ENU', 'gyrZ': 0.005008840002119541, 'gyrX': -0.0024311246816068888, 'gyrY': 0.003479980630800128}}
{'Acceleration': {'Delta v.z': 0.09783565998077393, 'Delta v.y': -0.000495612621307373, 'Delta v.x': 0.0016425251960754395, 'frame': 'ENU', 'accY': -0.049656979739665985, 'accX': 0.16383680701255798, 'accZ': 9.783573150634766}, 'Timestamp': {'PacketCounter': 209, 'SampleTimeCoarse': 2, 'SampleTimeFine': 21137}, 'Magnetic': {'magZ': -0.8042738437652588, 'frame': 'ENU', 'magX': -0.005401611328125, 'magY': 0.521867036819458}, 'Angular Velocity': {'Delta q1': -4.3567270040512085e-06, 'Delta q0': 1.0000001192092896, 'Delta q3': 3.1896866858005524e-05, 'Delta q2': 2.108421358570922e-05, 'frame': 'ENU', 'gyrZ': 0.006379373837262392, 'gyrX': -0.0008713454590179026, 'gyrY': 0.004216842353343964}}
...

Hope that helps :)

fcolas, 2017-02-02:
Dear Kai,
Thanks for the additional tests.
What I can say:
1. I have trouble understanding the output given that `GetOptionFlags` exists and is actually properly called in your test with your other device.

2. Ok

3. Commenting out the general call to the inspection doesn't help us know which messages are properly supported by your MTw device. Can you continue commenting one by one the calls that cause issue?

4. It's nice to see, at least.

Thanks

metzge-ka, 2017-02-03:
Hey Francis,

1)      I don’t know what to comment out next, after commenting out #1519 I commented out #1520 and #1430 (the latter doesn’t output anything), as well as commenting out #1704…
Sorry for my noobieness …

2)      Can I do something about the GetOptionFlags issue?
Will gladly continue with this but I have no clue, unfortunately…

Cheers,
Kai

fcolas, 2017-02-03:
Dear Kai,

For the `GetOptionFlags` issue, I suspect that you did something more than just commenting line 1519 (when you make and unmake several changes, it can happen at some point to forget exactly what was done). To make sure, just revert all your changes (look at git documentation if you're not sure how) and comment again line 1519.

As for what to comment out next, it's written in the traceback of the full output (you truncated it but it would have been 1522). The only lines that are relevant to comment out at this debugging stage of your device are between 1519 and 1539 (but the point is not to comment them all a priori but to know which one need to be commented out).

Cheers,

metzge-ka, 2017-02-27:
Dear Francis,
sorry for coming back that late...but I don't know what I can do to fix the other issue...

Now, I have another issue, though...it would be awesome if you could help me :-)
When I connect an MTi-G-700 sensor and run:
`pi@raspberrypi:~/catkin_ws/src/ethzasl_xsens_driver-master/nodes $ ./mtdevice.py -e`
I get the following output: 
_{'Status': {'StatusWord': 129}, 'Timestamp': {'PacketCounter': 321, 'SampleTimeFine': 64662}, 'Orientation Data': {'Yaw': 0.006209181156009436, 'Pitch': -0.04300256446003914, 'frame': 'ENU', 'Roll': 0.5484659075737}}
{'GNSS': {'numSvs': 11, 'iTOW': 6500, 'svs': [{'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}]}, 'Timestamp': {'PacketCounter': 322, 'SampleTimeFine': 2827657820L}}
{'Status': {'StatusWord': 129}, 'Timestamp': {'PacketCounter': 323, 'SampleTimeFine': 64762}, 'Orientation Data': {'Yaw': 0.006437348201870918, 'Pitch': -0.04260209575295448, 'frame': 'ENU', 'Roll': 0.5487979650497437}}
Traceback (most recent call last):
  File "./mtdevice.py", line 1704, in <module>
    main()
  File "./mtdevice.py", line 1470, in main
    print mt.read_measurement(mode, settings)
  File "./mtdevice.py", line 632, in read_measurement
    return self.parse_MTData2(data)
  File "./mtdevice.py", line 937, in parse_MTData2
    parse_GNSS(data_id, content, ffmt))
  File "./mtdevice.py", line 756, in parse_GNSS
    content)
ValueError: too many values to unpack_

Am I doing something wrong in the settings with the MTmanager? It doesn't matter what I enable with the sensor, it seems to have something to do with the GPS signal or packets...or more precise with the unpacking.

fcolas, 2017-02-27:
Dear Kai,

Please don't mix the issues: this one is about MTw2; if you have a problem with an MTi-G-700, you should open a new issue on github.
As for what to do for the current issue, it's written in my previous reply. If you have a specific question on what I suggest, don't hesitate.

Cheers,

metzge-ka, 2017-03-15:
The problem for the MTw2, when running pi@raspberrypi:~/catkin_ws/src/ethzasl_xsens_driver-master/nodes $ ./mtdevice.py -b 0 -v –i:
seems to be fixed now. 

pi@raspberrypi:~/catkin_ws/src/ethzasl_xsens_driver-master/nodes $ ./mtdevice.py -b 0 -v –i
Trying 115200 bd: MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [00 B4 16 28 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 B4 16 28 03 78 00 02 00 00 01 80 00 27 04 00 02 00 00 00]
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
ok.
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [00 B4 16 28 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 B4 16 28 03 78 00 02 00 00 01 80 00 27 04 00 02 00 00 00]
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x36 (MTData2) with 98 data bytes: [10 20 02 00 00 10 60 04 00 00 01 DE 10 70 04 00 00 00 00 40 20 0C 3C B6 55 67 BB 71 8C FC 3F C4 6B 43 40 10 0C 3A 8C 2A 00 B9 3F F0 00 3D 96 78 28 80 20 0C 3A F8 44 0D 3A 89 47 48 3A E2 27 F3 80 30 10 3F 80 00 00 38 3E 30 01 37 D2 54 00 38 2D 40 00 C0 20 0C BD 2C C8 80 BE F9 DA 30 BF DB 13 DE]
{'Acceleration': {'Delta v.z': 0.07347136735916138, 'Delta v.y': -0.0001830458641052246, 'Delta v.x': 0.0010693669319152832, 'frame': 'ENU', 'accY': -0.0036857714876532555, 'accX': 0.022257519885897636, 'accZ': 1.534523367881775}, 'Timestamp': {'PacketCounter': 0, 'SampleTimeCoarse': 0, 'SampleTimeFine': 478}, 'Magnetic': {'magZ': -1.7115437984466553, 'frame': 'ENU', 'magX': -0.04218339920043945, 'magY': -0.4879927635192871}, 'Angular Velocity': {'Delta q1': 4.5344237150857225e-05, 'Delta q0': 1.0, 'Delta q3': 4.13060188293457e-05, 'Delta q2': 2.5073066353797913e-05, 'frame': 'ENU', 'gyrZ': 0.001725433743558824, 'gyrX': 0.001894117915071547, 'gyrY': 0.001047351397573948}}
MT: Got message id 0x36 (MTData2) with 98 data bytes: [10 20 02 00 01 10 60 04 00 00 02 42 10 70 04 00 00 00 00 40 20 0C 3E 05 C6 C1 BC CC 98 92 41 1E 30 10 40 10 0C 3A AC 04 00 B9 88 40 00 3D CA 7A E8 80 20 0C 3C 2A 31 8A 3B C4 C5 70 3C 35 31 2B 80 30 10 3F 80 00 01 38 59 D9 01 37 FB DE 00 38 67 ED 00 C0 20 0C BD 48 9A 00 BE F6 8C B0 BF DB 68 02]
{'Acceleration': {'Delta v.z': 0.09886723756790161, 'Delta v.y': -0.0002598762512207031, 'Delta v.x': 0.0013123750686645508, 'frame': 'ENU', 'accY': -0.024975094944238663, 'accX': 0.13064099848270416, 'accZ': 9.886734008789062}, 'Timestamp': {'PacketCounter': 1, 'SampleTimeCoarse': 0, 'SampleTimeFine': 578}, 'Magnetic': {'magZ': -1.714111566543579, 'frame': 'ENU', 'magX': -0.04897499084472656, 'magY': -0.48154211044311523}, 'Angular Velocity': {'Delta q1': 5.193893230170943e-05, 'Delta q0': 1.0000001192092896, 'Delta q3': 5.5295415222644806e-05, 'Delta q2': 3.0024908483028412e-05, 'frame': 'ENU', 'gyrZ': 0.011059085838496685, 'gyrX': 0.010387787595391273, 'gyrY': 0.006004981696605682}}
MT: Got message id 0x36 (MTData2) with 98 data bytes: [10 20 02 00 02 10 60 04 00 00 02 A6 10 70 04 00 00 00 00 40 20 0C 3E 13 6E 07 BC 3E DF 78 41 1E 61 A3 40 10 0C 3A BD 46 00 B9 05 60 00 3D CA BA 60 80 20 0C 3C 33 6F 29 3B 8D DD 30 3C 23 F3 17 80 30 10 3F 80 00 01 38 65 AD 00 37 B5 96 01 38 51 DB 00 C0 20 0C BD 35 59 80 BE FA F8 40 BF DB 12 5C]

It works just as well as the MTi sensor :-)
Thanks a lot!

fcolas, 2017-03-15:
Well, I'm sorry but this doesn't look like the output of the inspection action but it seems like, for some reason, you used a n-dash `–` character instead of the standard dash `-` (see, it's not the same length).
Can you try again with the proper dash?

(I didn't fix anything because I'm without diagnostics I cannot, which is why there's no real reason that it'd be fixed.)

fcolas, 2017-11-28:
It should be fixed in #60

fcolas, 2018-04-04:
@metzge-ka Can you confirmed it's fixed?

closed 2018-07-16 by fcolas