ethz-asl/ethzasl_xsens_driver #33: Inspection of MTi-30 crashes

ipa-mig, 2017-01-31:
I'm not quite sure if this is related to #32 

Setup:
- Ubuntu 16.04.1 all up-to-date
- ROS Kinetic
- `xsens_driver` from apt: 2.0.1
- HW: MTi-30-2A5G4 over USB

When calling:

```
rosrun xsens_driver mtdevice.py -iv
```

the script crashes with the following output:

```
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 08 A6 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 08 A6 00 00 00 00 00 00 00 00 00 27 01 04 0C 49 04 00]
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  MT: Write message id 0x00 (ReqDID) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x01 (DeviceID) with 4 data bytes: [03 60 08 A6]
0x036008A6
  product code:  MT: Write message id 0x1C (ReqProductCode) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x1D (ProductCode) with 20 data bytes: [4D 54 69 2D 33 30 2D 32 41 35 47 34 20 20 20 20 20 20 20 20]
'MTi-30-2A5G4'
  firmware revision:  MT: Write message id 0x12 (ReqFWRev) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x13 (FirmwareRev) with 11 data bytes: [01 04 0C 00 00 00 0D 00 00 D5 76]
(1, 4, 12)
  baudrate:  MT: Write message id 0x18 (SetBaudrate) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x19 (SetBaudrateAck) with 1 data bytes: [02]
2
  error mode:  MT: Write message id 0xDA (SetErrorMode) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x42 (Error) with 5 data bytes: [04 00 00 00 00]
message unsupported by your device.
  option flags:  MT: Write message id 0x48 (SetOptionFlags) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x49 (SetOptionFlagsAck) with 4 data bytes: [00 00 00 00]
Traceback (most recent call last):
  File "/home/mig/git/sandbox_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1704, in <module>
    main()
  File "/home/mig/git/sandbox_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1430, in main
    inspect(mt, device, baudrate)
  File "/home/mig/git/sandbox_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1520, in inspect
    try_message("option flags:", mt.GetOptionFlags, hex_fmt(8))
  File "/home/mig/git/sandbox_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1506, in try_message
    print formater(f(*args, **kwargs))
  File "/home/mig/git/sandbox_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 288, in GetOptionFlags
    set_flags, clear_flags = struct.unpack('!II', data)
struct.error: unpack requires a string argument of length 8
```

Commenting out line [1520](https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtdevice.py#L1520) does the trick and it works perfectly.

```
rosrun xsens_driver mtdevice.py -e
rosrun xsens_driver mtnode.py
```
without any parameters set work fine (even though the node somtimes complains about `Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.` when echoing the topics)

fcolas, 2017-02-02:
Ok, interesting. The documentation of this message was a bit unclear.
I've pushed the branch `fix_get_option_flags` can you try it out and tell me whether it's ok now?

ipa-mig, 2017-02-02:
Perfect. all good. Thank you for the fix.

For reference, below the output for `rosrun xsens_driver mtdevice.py -iv`

```
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 08 A6 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 08 A6 00 00 00 00 00 00 00 00 00 27 01 04 0C 49 04 00]
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  MT: Write message id 0x00 (ReqDID) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x01 (DeviceID) with 4 data bytes: [03 60 08 A6]
0x036008A6
  product code:  MT: Write message id 0x1C (ReqProductCode) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x1D (ProductCode) with 20 data bytes: [4D 54 69 2D 33 30 2D 32 41 35 47 34 20 20 20 20 20 20 20 20]
'MTi-30-2A5G4'
  firmware revision:  MT: Write message id 0x12 (ReqFWRev) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x13 (FirmwareRev) with 11 data bytes: [01 04 0C 00 00 00 0D 00 00 D5 76]
(1, 4, 12)
  baudrate:  MT: Write message id 0x18 (SetBaudrate) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x19 (SetBaudrateAck) with 1 data bytes: [02]
2
  error mode:  MT: Write message id 0xDA (SetErrorMode) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x42 (Error) with 5 data bytes: [04 00 00 00 00]
message unsupported by your device.
  option flags:  MT: Write message id 0x48 (SetOptionFlags) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x49 (SetOptionFlagsAck) with 4 data bytes: [00 00 00 00]
0x00000000
  location ID:  MT: Write message id 0x84 (SetLocationID) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x85 (SetLocationIDAck) with 2 data bytes: [00 00]
0x0000
  transmit delay:  MT: Write message id 0xDC (SetTransmitDelay) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x42 (Error) with 5 data bytes: [04 00 00 00 00]
message unsupported by your device.
  synchronization settings:  MT: Write message id 0x2C (SetSyncSettings) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x2D (SetSyncSettingsAck) with 0 data bytes: []
[]
  general configuration:  MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 60 08 A6 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 60 08 A6 00 00 00 00 00 00 00 00 00 27 01 04 0C 49 04 00]
{   'Master device ID': 56625318,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 56625318,
    'length': 0,
    'number of devices': 1,
    'output-mode': 0,
    'output-settings': 0,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  MT: Write message id 0xC0 (SetOutputConfiguration) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0xC1 (SetOutputConfigurationAck) with 40 data bytes: [10 20 FF FF 10 60 FF FF 20 10 01 90 40 20 01 90 40 10 01 90 80 20 01 90 80 30 01 90 C0 20 00 64 08 10 01 90 E0 20 FF FF]
[(0x1020, 65535), (0x1060, 65535), (0x2010, 400), (0x4020, 400), (0x4010, 400), (0x8020, 400), (0x8030, 400), (0xC020, 100), (0x0810, 400), (0xE020, 65535)]
  string output type:  MT: Write message id 0x8E (SetStringOutputType) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x8F (SetStringOutputTypeAck) with 2 data bytes: [00 00]
0
  period:  MT: Write message id 0x04 (SetPeriod) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
[... 17 lines cut ...]
0x0000
  output settings:  MT: Write message id 0xD2 (SetOutputSettings) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0xD3 (SetOutputSettingsAck) with 4 data bytes: [00 00 00 00]
0x00000000
  GPS coordinates (lat, lon, alt):  MT: Write message id 0x6E (SetLatLonAlt) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x6F (SetLatLonAltAck) with 24 data bytes: [40 4A 1E C8 6E 8C 85 64 40 1B 5A 55 7F 4C D1 61 00 00 00 00 00 00 00 00]
(52.24049169408207, 6.838216771186439, 0.0)
  available scenarios:  MT: Write message id 0x62 (ReqAvailableScenarios) with 0 data bytes: []
MT: Got message id 0x63 (AvailableScenarios) with 110 data bytes: [27 08 67 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20 20 20 20 20 28 08 68 69 67 68 5F 6D 61 67 5F 64 65 70 20 20 20 20 20 20 20 20 29 08 64 79 6E 61 6D 69 63 20 20 20 20 20 20 20 20 20 20 20 20 20 2A 08 6C 6F 77 5F 6D 61 67 5F 64 65 70 20 20 20 20 20 20 20 20 20 2B 0C 76 72 75 5F 67 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20]
[   (39, 8, 'general'),
    (40, 8, 'high_mag_dep'),
    (41, 8, 'dynamic'),
[... cut ...] , 'low_mag_dep'),
    (43, 12, 'vru_general')]
  current scenario ID:  MT: Write message id 0x64 (SetCurrentScenario) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x65 (SetCurrentScenarioAck) with 2 data bytes: [00 27]
39
  UTC time:  MT: Write message id 0x60 (SetUTCTime) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x61 (UTCTime) with 12 data bytes: [0C 13 6E 40 07 B2 01 01 00 00 2C 00]
(202600000, 1970, 1, 1, 0, 0, 44, 0)
```

closed 2017-02-02 by fcolas

fcolas, 2017-02-02:
Thanks for the confirmation. It shouldn't break anything, but I couldn't test it with the device I have.

PR #34 (2017-02-02): fix GetOptionFlags
fix #33