ethz-asl/ethzasl_xsens_driver #36: Problem with MTi-G-700

metzge-ka, 2017-02-27:
Dear Francis,
sorry for coming back that late...but I don't know what I can do to fix the other issue...

Now, I have another issue, though...it would be awesome if you could help me :-)
When I connect an MTi-G-700 sensor and run:
`pi@raspberrypi:~/catkin_ws/src/ethzasl_xsens_driver-master/nodes $ ./mtdevice.py -e`
I get the following output: 
_{'Status': {'StatusWord': 129}, 'Timestamp': {'PacketCounter': 321, 'SampleTimeFine': 64662}, 'Orientation Data': {'Yaw': 0.006209181156009436, 'Pitch': -0.04300256446003914, 'frame': 'ENU', 'Roll': 0.5484659075737}}
{'GNSS': {'numSvs': 11, 'iTOW': 6500, 'svs': [{'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}, {'gnssId': 1, 'cno': 45, 'svId': 147, 'flags': 35}]}, 'Timestamp': {'PacketCounter': 322, 'SampleTimeFine': 2827657820L}}
{'Status': {'StatusWord': 129}, 'Timestamp': {'PacketCounter': 323, 'SampleTimeFine': 64762}, 'Orientation Data': {'Yaw': 0.006437348201870918, 'Pitch': -0.04260209575295448, 'frame': 'ENU', 'Roll': 0.5487979650497437}}
Traceback (most recent call last):
  File "./mtdevice.py", line 1704, in <module>
    main()
  File "./mtdevice.py", line 1470, in main
    print mt.read_measurement(mode, settings)
  File "./mtdevice.py", line 632, in read_measurement
    return self.parse_MTData2(data)
  File "./mtdevice.py", line 937, in parse_MTData2
    parse_GNSS(data_id, content, ffmt))
  File "./mtdevice.py", line 756, in parse_GNSS
    content)
ValueError: too many values to unpack_

Am I doing something wrong in the settings with the MTmanager? It doesn't matter what I enable with the sensor, it seems to have something to do with the GPS signal or packets...or more precise with the unpacking. 

If I run:
`pi@raspberrypi:~/catkin_ws/src/ethzasl_xsens_driver-master/nodes $ ./mtdevice.py -b 0 -v -i`
_Trying 115200 bd: MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [07 70 04 5E 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 07 70 04 5E 00 00 00 00 00 00 00 01 00 01 01 04 05 37 03 00]
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x10 (GoToMeasurement) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x11 (GoToMeasurementAck) with 0 data bytes: []
ok.
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
[... 74 lines cut ...]
0x0000
  output settings:  MT: Write message id 0xD2 (SetOutputSettings) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0xD3 (SetOutputSettingsAck) with 4 data bytes: [00 00 00 01]
0x00000001
  GPS coordinates (lat, lon, alt):  MT: Write message id 0x6E (SetLatLonAlt) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x6F (SetLatLonAltAck) with 24 data bytes: [40 48 0A DB 60 92 81 D5 40 26 8E 1E B5 31 42 7C 40 83 8D B4 48 EC FC 00]
(48.08481986192388, 11.277578031790874, 625.7130297197727)
  available scenarios:  MT: Write message id 0x62 (ReqAvailableScenarios) with 0 data bytes: []
MT: Got message id 0x63 (AvailableScenarios) with 110 data bytes: [01 07 47 65 6E 65 72 61 6C 20 20 20 20 20 20 20 20 20 20 20 20 20 02 07 47 65 6E 65 72 61 6C 4E 6F 42 61 72 6F 20 20 20 20 20 20 20 03 07 47 65 6E 65 72 61 6C 4D 61 67 20 20 20 20 20 20 20 20 20 20 04 07 41 75 74 6F 6D 6F 74 69 76 65 20 20 20 20 20 20 20 20 20 20 05 07 41 75 74 6F 55 72 62 61 6E 43 61 6E 79 6F 6E 20 20 20 20 20]
[   (1, 7, 'General'),
    (2, 7, 'GeneralNoBaro'),
    (3, 7, 'GeneralMag'),
    (4, 7, 'Automotive'),
    (5, 7, 'AutoUrbanCanyon')]
  current scenario ID:  MT: Write message id 0x64 (SetCurrentScenario) with 0 data bytes: []
MT: Got message id 0x65 (SetCurrentScenarioAck) with 2 data bytes: [00 01]
1
  UTC time:  MT: Write message id 0x60 (SetUTCTime) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x61 (UTCTime) with 12 data bytes: [1F 58 98 E0 07 E1 02 1B 0A 04 1A 00]
(525900000, 2017, 2, 27, 10, 4, 26, 0)_

This looks fine for me ...

fcolas, 2017-02-27:
Thanks for the report.
I guess I see the problem. Can you checkout the branch `fix_gnss_pvt_parsing` and tell me if it's better?

metzge-ka, 2017-03-07:
Thanks, but I am not sure what to do exaclty. 
What do you mean by checkout the branch fix_gnss_pvt_parsing?
(I cannot find a variable with that name...)
Do you mean the method parse_GNSS()?

fcolas, 2017-03-07:
Well, a branch is a git concept you might want to learn about.
Here is a good introductory book: https://git-scm.com/book/en/v2

hlware, 2017-03-09:
I had the same Problem with unpacking GNSS messages.  The branch fix_gnss_pvt_parsing solved that unpacking problem, with additionally correcting line 767 `o['bdop'] *= 0.01` to `o['ndop'] *= 0.01`.
However this fix directly results in the next error
`File ".../mtnode.py", line 508, in fill_from_GNSS
  self.gps_msg.status.service = NavSatStatus.SERVICE_GPS
AttributeError: 'int' object has no attributee 'service'`

fcolas, 2017-03-09:
Thanks for the report. I supposedly fixed both issues: can you confirm?

hlware, 2017-03-09:
Thanks a lot, that solved the problem.

fcolas, 2017-03-09:
Great, I'll merge it in.

PR #37 (2017-03-09): Fix gnss pvt parsing
Fix #36
closed 2017-03-09 by fcolas