ethz-asl/ethzasl_xsens_driver #40: ROS Xsens driver error with MTi-G-700

Seong-konkukUniversity, 2017-04-07:
Hi Dear,

I have a problem with ROS Xsens dirver.

As shown in the attached files, When I  parsed the GPSSOL data from MTi-G-700, the data length is 52 byte which is not corresponding to the data length in MT Low-Level Communication Protocol Documentation. So I add the 'xxx' in the parsing code. Then I obtained the ECEF position. 
However the other data such as ECEF velocity, position accuracy, etc. 
I hope you to help me to get the whole data from GPSSOL message.

additionally, I want to know the mean of 'x' in the function "struct.unpack()

Thanks for your help.

[image]
[image]

fcolas, 2017-04-07:
Hi,
I won't say better what `x` means in `struct.unpack` than the [documentation](https://docs.python.org/2/library/struct.html#format-characters).
The problem with adding `x` so as to make the size fit is that you're never sure you're adding it to the correct place: the data you get out can then be completely wrong.
Can you please paste the output of the command `./mtdevice.py -i`?
With this information, we can contact xsens to try to get an answer.

As for the values you show it doesn't seem like you got a GPS fix.

fcolas, 2017-05-19:
No update on this?

fcolas, 2018-04-04:
@Seong-konkukUniversity This data type has been deprecated. It would be nice to have a solution for this issue but without your help, I'll have to close it.