ethz-asl/ethzasl_xsens_driver #44: Extension for Synchronization

raymond30031, 2017-05-19:
Hello,

Is there plan to add commands for setting the synchronization settings of the IV generation devices?

If there is no plan in the near future and I wish to contribute, is it better to start the discussion here or through email?

Regards,
Juichung Kuo

fcolas, 2017-05-19:
Hi,
There is already code in `mtdevice.py` to get or set the synchronization settings (which can therefore be used in an interactive python shell) so I guess you mean adding command line arguments to set this configuration in a more user-friendly manner, am I right?
I didn't have specific plans as I didn't look too much at those settings, but if you already have an idea, we could indeed discuss it here and see how to implement it.
Best,

raymond30031, 2017-05-19:
Hi,

Yeah, that is what I meant. I will branch off and I am sure I will have more questions and comments once I start working on it.

- JC

raymond30031, 2017-05-21:
Hello Francis,

I just started adding some command options for synchronization.
I have created a separate branch, but I don't have write permission for push.
Could i get such permission?

I would say I am a junior programmer, so I am open to suggestions.
I was going to have every setting argument as a number, but I realize it might be difficult to parse them.
I am still thinking what is the best way of generating commands for the following options:

Synchronization settings:
    The format follows the xsens protocol documentation:
    |Function| Line| Trigger Type| Skip First|...
    Skip Factor| Pulse Width| Delay or Clock period or offset|

        Function (see manual for details):
            3 Trigger indication
            4 Interval Transition Measurement
            8 SendLatest
            9 ClockBiasEstimation
            11 StartSampling
        Line (manual for details):
            0 ClockIn
            1 GPSClockIn (only available for 700/710)
            2 Input Line (SyncIn)
            4 SyncOut
            5 ExtTimepulseIn (only available for 700/710)
            6 Software (only available for SendLatest with ReqData message)
        Polarity:
            1 Positive pulse/ Rising edge
            2 Negative pulse/ Falling edge
            3 Both/ Toggle
        Trigger Type:
            0 multiple times
            1 once
        Skip First (unsigned_int):
            Number of initial events to skip before taking actions
        Skip Factor (unsigned_int):
            Number of events to skip before taking action again
            Ignored with ReqData.
        Pulse Width (unsigned_int):
            Ignored for SyncIn.
            For SyncOut, the width of the generated pulse in 100 microseconds
            unit. Ignored for Toggle pulses.
        Delay:
            Delay after receiving a sync pulse to taking action,
            100 microseconds units, range [0...600000]
        Clock Period:
            Reference clock period in milliseconds for ClockBiasEstimation
        Offset:
            Offset from event to pulse generation.
            100 microseconds unit, range [-30000...+30000]

Cheers,
JC

raymond30031, 2017-05-27:
Hello @fcolas 

I have finished adding command line support for set synchronization settings and UTC timer, but I need permission to send a PR.

-JC

raymond30031, 2017-06-16:
Hello @fcolas 

My apologies, this is my first time contributing. I was branching off a clone of this repo instead of forking.
I have created a PR for this enhancement.
I look forward to your feedbacks.

JC

fcolas, 2017-07-17:
Done in #46

closed 2017-07-17 by fcolas