ethz-asl/ethzasl_xsens_driver #45: Euler and Quaternions

veronicagruning, 2017-05-19:
So when running this node, I want both the Euler and Quaternion angles, so within the function starting at Ln 297 and the again at Ln 421 I wanted to include:
roll, pitch, yaw = euler_from_quaternion(
                degree(orient_data['x']), degree(orient_data['y']),
                degree(orient_data['z']), degree(orient_data['w'])
After if/else and try/except, but syntax errors for the next function keep appearing. 
  File "mtnode.py", line 322
    def fill_from_Auxiliary(aux_data):
      ^
SyntaxError: invalid syntax

Any ideas how I can get both Euler and Quaternion?

fcolas, 2017-05-22:
It is difficult to debug a SyntaxError without seeing the code but that might not be the problem.

I'm not exactly sure what you really want. I believe mark iv devices can be set to output their orientation both in quaternion and Euler angles. But the standard representation for orientations in ROS is quaternions for good theoretical reasons and we probably don't want to change that in a generic node (notice how there is not a Euler angle message type in `geometry_msgs`).

So my question is more whether you want to have both informations directly from the IMU, or if you just want to have a topic with Euler angles.
If you want to access the Euler angle output by the IMU, maybe the simplest is to read and parse the `imu_data_str` topic, which publishes in string form the content of the messages received by the IMU.
If you just want a topic with Euler angles, the simplest would be to create a small node, listening to the `imu/data` topic, to do the conversion.
Does that answer your question?

veronicagruning, 2017-05-22:
When running the node as it is currently, you see the quaternions live in the terminal and I want to see them concurrently. An engineer from xsens said there should be no issue in having both data strings being outputted. Ideally both information will come from the IMU directly.

fcolas, 2017-05-22:
I'm not sure what you're talking about since the node (`mtnode.py`) shouldn't output data in the terminal (if it is, please file in a bug report). 
You might be referring to the command line configuration and inspection tool: `mtdevice.py` which is not a ROS node and not intended to by used as such during an application.
If you're unclear, please read the [documentation](http://wiki.ros.org/xsens_driver).

Beside this, the engineer from xsens is right in that you can setup your imu (mark iv) so that it streams the data as both quaternion and Euler angles. But then the imu driver node `mtnode.py` does not publish orientation as Euler angles so this redundant information is discarded. If you have a good reason to want both, the solution I talked about was for you to look at the `imu_data_str` topic which conveniently publishes a textual representation of the data streamed by the IMU (similar to what `mtdevice.py` shows on the command line with the `-e` option).

(But honestly, I still don't see why you want your IMU to stream this information under both formats: it seems way easier to listen to the quaternion and convert it to the specific angles you might need.)

fcolas, 2017-11-28:
Closing since it doesn't really seems to be an issue and you don't seem to be interested any more.

closed 2017-11-28 by fcolas