ethz-asl/ethzasl_xsens_driver #47: No angular velocity data

spacemanspiff0211, 2017-06-29:
Hey,

First of all thanks for the drivers. I am using an MTi-10 and when I run the `mtnode.py` node i don't seem to get the angular velocity data. But when i run `rosrun xsens_driver mtdevice.py -m 2 -f 10` I seem to get angular velocity values. The output of the console is 

```
started roslaunch server http://tegra-ubuntu:33887/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /xsens_driver/baudrate: 115200
 * /xsens_driver/device: /dev/xsens
 * /xsens_driver/frame_id: xsens
 * /xsens_driver/frame_local: ENU
 * /xsens_driver/no_rotation_duration: 0
 * /xsens_driver/timeout: 0.002

NODES
  /
    xsens_driver (xsens_driver/mtnode.py)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[xsens_driver-1]: started with pid [11925]
[INFO] [1498763213.397881]: Found parameter: ~device, value: /dev/xsens
[INFO] [1498763213.401874]: Found parameter: ~baudrate, value: 115200
[INFO] [1498763213.406158]: Found parameter: ~timeout, value: 0.002
[INFO] [1498763213.407166]: MT node interface: /dev/xsens at 115200 bd.
[INFO] [1498763213.456293]: Found parameter: ~no_rotation_duration, value: 0
[INFO] [1498763213.459914]: Found parameter: ~frame_id, value: xsens
[INFO] [1498763213.463558]: Found parameter: ~frame_local, value: ENU
[WARN] [1498763229.732860]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
^C[xsens_driver-1] killing on exit
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/xsens_driver/mtnode.py", line 726, in <module>
    main()
  File "/opt/ros/kinetic/lib/xsens_driver/mtnode.py", line 722, in main
    driver.spin()
  File "/opt/ros/kinetic/lib/xsens_driver/mtnode.py", line 131, in spin
    self.spin_once()
  File "/opt/ros/kinetic/lib/xsens_driver/mtnode.py", line 638, in spin_once
    data = self.mt.read_measurement()
  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 628, in read_measurement
    mid, data = self.read_msg()
  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 131, in read_msg
    if ord(self.waitfor()) != 0xFA:
  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 81, in waitfor
    buf.extend(self.device.read(size-len(buf)))
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 511, in read
    raise SerialException('read failed: %s' % (e,))
serial.serialutil.SerialException: read failed: (4, 'Interrupted system call')
shutting down processing monitor...
... shutting down processing monitor complete
done
```
 and the rostopic echo message is 
```
---
header: 
  seq: 894
  stamp: 
    secs: 1498763229
    nsecs: 714687108
  frame_id: xsens
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]
linear_acceleration: 
  x: -0.000661169760861
  y: 0.00464787147939
  z: 0.0979677215219
linear_acceleration_covariance: [0.0004, 0.0, 0.0, 0.0, 0.0004, 0.0, 0.0, 0.0, 0.0004]
---

```
Any help is appreciated.

Thanks

fcolas, 2017-06-30:
It does seem like a configuration issue. The output from the node really looks like it only receives accelerations.
Can you paste the output of both `rosrun xsens_driver mtdevice.py -i` and `rosrun xsens_driver mtdevice.py -e`?

By the way, `rosrun xsens_driver mtdevice.py -m 2 -f 10` is not changing any configuration or anything: the `-m` and `-f` options only make sense with the `-l`/`--legacy-configure` command. `rosrun xsens_driver mtdevice.py -m 2 -f 10` is only doing the default command, which is echoing the messages, and your flags are ignored (see usage if unsure).

spacemanspiff0211, 2017-06-30:
Hey,
by running `rosrun xsens_driver mtdevice.py -m 2 -f 10` i was using the test command from official ros driver from Xsens.
Anyways the output of `rosrun xsens_driver mtdevice.py -i` is 
```
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x0160025E
  product code: 'MTi-10-2A5G4'
  firmware revision: (1, 5, 1)
  baudrate: 2
  error mode:  message unsupported by your device.
  option flags:  0x00000000
  location ID:  0x0000
  transmit delay:  message unsupported by your device.
  synchronization settings:  []
  general configuration: {   'Master device ID': 23069278,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 23069278,
    'length': 0,
    'number of devices': 1,
    'output-mode': 0,
    'output-settings': 0,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  [(0x2010, 100), (0x3010, 50), (0x4020, 100), (0x8020, 100), (0xC020, 100), (0xE020, 65535), (0x1060, 65535)]
  string output type: 0
  period: 1152
  alignment rotation sensor: (1.0, 0.0, 0.0, 0.0)
  alignment rotation local: (1.0, 0.0, 0.0, 0.0)
  output mode:  0x0000
  extended output mode:  0x0000
  output settings:  0x00000000
  GPS coordinates (lat, lon, alt): (52.24049169408207, 6.838216771186439, 0.0)
  available scenarios:  message unsupported by your device.
  current scenario ID:  message unsupported by your device.
  UTC time: (227100000, 1970, 1, 1, 4, 45, 53, 0)
```
The o/p of `rosrun xsens_driver mtdevice.py -e` is
```
{'Acceleration': {'frame': 'ENU', 'accY': 0.13117225468158722, 'accX': 1.0620391368865967, 'accZ': 7.254292011260986}, 'Status': {'StatusWord': 5}, 'Magnetic': {'magZ': -0.6732985973358154, 'frame': 'ENU', 'magX': -0.18708159029483795, 'magY': -0.16067351400852203}, 'Angular Velocity': {'frame': 'ENU', 'gyrZ': -0.001104344497434795, 'gyrX': -0.006229492370039225, 'gyrY': 0.0022216620855033398}, 'Timestamp': {'SampleTimeFine': 171382236}}
```
Thanks

fcolas, 2017-07-03:
Hi,
There is a small confusion: the official Xsens driver for ROS is not this one. A few years ago, XSens took this code, modified it and made that their official ROS driver (it's ok, it's a permissive open source licence). But I continued working on my version every now and then so the code as diverged: make sure you're using my version of the code and don't mix command line syntax (if you prefer, you can use theirs but then I can't offer any help).

Judging from the configuration and the output you show, indeed, there should be more in the messages.
Can you show the list of topics published: `$ rostopic list`
And can you echo the following topics: `imu/data`, `velocity`, `imu/mag`

spacemanspiff0211, 2017-07-07:
Hey,
For some unknown reason, it started working again. So thanks for all your support. I'm going to mark this issue closed.

fcolas, 2017-07-10:
Hi,
That's great; don't hesitate to ask if something weird happens again.
It might simply have been a configuration issue that you fixed somehow in the process of submitting the ticket.

closed 2017-07-10 by fcolas