ethz-asl/ethzasl_xsens_driver #51: No orientation for MTw2 devices

rastaxe, 2017-07-24:
First of all, I want to thank you for this repository. I tried the code from xsens for MTw devices and nothing works! But this code seems to work somehow, although there is something different for MTw2 devices.
Running inspect I got:
```
./mtdevice.py -i

Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x00B4126B
  product code: 'MTW2-3A7G6'
  firmware revision: (4, 0, 2)
  baudrate: 2
  error mode:  MTException: Timeout: waiting for message
```

If I run the node, on the /imu/data there is not orientation data, while the other fields contain nans or close to zero values. This is an output of ./mtdevice -e:

`{'Calib': {'magZ': nan, 'magY': 6.310593491255738e-40, 'magX': nan, 'frame': 'ENU', 'accY': nan, 'accX': nan, 'accZ': 1.4695075078549061e-38, 'gyrZ': nan, 'gyrX': nan, 'gyrY': 1.4158719683537952e-41}}'`

Can you help me with this type of device?

rastaxe, 2017-07-24:
Just an update. I used the mtmanager program from xsens to configure the device with the XDA processing preset, all the data are in 32-bit floating point. Now, when I start the ros node, it prints continuously some data
`(0.000338078, 0.009810718332989691, 102.0)`
`(-0.0059709992468379575, 0.009101840497048295, -0.010500355284005785)`
`(0.000298977, 0.009328674340196078, 107.0)`
`(-0.006394836279216699, 0.008683433485913532, -0.010023346822737837)`
`(0.00029397, 0.008893203172897195, 112.0)`
`(-0.006404935544333341, 0.006039231290238613, -0.007619916477503733)
`

The echo on mtdevice is now:

`{'Acceleration': {'frame': 'ENU', 'Delta v.z': 0.09914320707321167, 'Delta v.y': -0.0006480216979980469, 'Delta v.x': -0.0008548498153686523}, 'Status': {'StatusWord': 1}, 'Temperature': {'Temp': 32.515625}, 'Timestamp': {'PacketCounter': 63, 'SampleTimeFine': 6816}, 'Magnetic': {'magZ': -0.87723708152771, 'frame': 'ENU', 'magX': 0.14348530769348145, 'magY': 0.45084524154663086}, 'Angular Velocity': {'Delta q1': -3.815256059169769e-05, 'Delta q0': 1.0000001192092896, 'frame': 'ENU', 'Delta q2': 5.028117811889388e-05, 'Delta q3': -4.1328370571136475e-05}, 'Pressure': {'Pressure': 100063}}
{'Acceleration': {'frame': 'ENU', 'Delta v.z': 0.09965825080871582, 'Delta v.y': -0.0008144974708557129, 'Delta v.x': -0.0008847713470458984}, 'Status': {'StatusWord': 1}, 'Temperature': {'Temp': 32.515625}, 'Timestamp': {'PacketCounter': 64, 'SampleTimeFine': 6916}, 'Magnetic': {'magZ': -0.8797299861907959, 'frame': 'ENU', 'magX': 0.14358782768249512, 'magY': 0.44843316078186035}, 'Angular Velocity': {'Delta q1': -3.0681490898132324e-05, 'Delta q0': 1.0000001192092896, 'frame': 'ENU', 'Delta q2': 4.341546446084976e-05, 'Delta q3': -5.0144270062446594e-05}}
`

However, the orientation in the /imu/data topic is still zero.
Do you know how I can retrieve the orientation data?

fcolas, 2017-08-18:
Hi,
The inspection problem is related to issue #32 but I couldn't get enough information for debugging. At that time, we realized that the MTw2 didn't answer to some messages it didn't know instead of the documented behaviour which is to answer with an error message.
I wanted to make sure which where the messages it didn't correctly answer. For that, I'd like you to try the inspection by commenting out the specific lines for which you get a timeout inside the [inspection function](https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtdevice.py#L1636-L1661). If you could do that, showing the result of the final inspection and telling me which lines you had to comment, it would be great (and it would help me helping you).

As for your second message, there where some `print` lines for debugging that slipped into the last commit; pull again, it should be fixed now.
The `echo` output seems to indicate that your IMU is not currently configured to output the orientation. If you help me solve the `inspect` problem, we should be able to make sure of that and attempt to use `mtdevice.py` to configure the IMU in order to give the orientation.

rastaxe, 2017-09-04:
Hi, 
  finally, I was able to successfully use the sensor, using the xsensdeviceapi library. However, when I have time I will help you to solve the inspect problem.

fcolas, 2017-11-28:
I'm going to close this issue since you found a solution. The `inspect` problem is supposed to be fixed in `fix_mt1_series_inspect` branch (see #60) if you can help test it, it would be great.

closed 2017-11-28 by fcolas

guilhermeguimaraes, 2017-11-28:
I also have issues with the driver. does not work properly

guilhermeguimaraes, 2017-11-28:
*fix_mt1_series_inspection

guilhermeguimaraes, 2017-11-28:
I am using a XSens MT1-1s IMU with Ubuntu 16.04 and ROS Kinetic

fcolas, 2017-11-30:
@guilhermeguimaraes Thanks for your interest in my node. However, this issue is closed and is not the same as yours so there's no need to reopen it.
Also, a few things to make both our lives easier:
- don't spread your messages across several issues: you'll need to repeat and I won't keep up with what you said where and I'll miss information;
- it's not a chat; take the time to build your question/answer/comment and don't comment several times in a row : just edit your previous message if you forgot something;
- use some [formatting capabilities](https://guides.github.com/features/mastering-markdown/), especially backticks (`) for error messages.

If in doubt, there are plenty of resources online for you to check how to file an efficient github issue that will allow me to better get your issue and hopefully help you faster.

madmage, 2018-02-06:
So, @fcolas , I see that @rastaxe was able to fix the 0 0 0 0 orientation problem, but how? I am having the same issue after upgrading the firmware of my Xsens.

fcolas, 2018-03-06:
@madmage I believe it was simply a configuration issue: try looking at the current configuration of your IMU to check what it should output.

Voidminded, 2018-05-08:
Is this issue closed without any fix?
@fcolas I believe it was not a configuration issue, as @rastaxe mentioned he gave up using this driver and is using the `xsensdeviceapi` library (from xsens) to get the orientation for MTw2.

fcolas, 2018-05-22:
@Voidminded  Well, I put the “need more info” tag on this question for a reason: only with the help of @rastaxe can we see what's the problem (and who's right) on this particular case. I'm willing to discuss this case to improve the driver, but only on stronger bases than our respective opinions.
However, if you have an issue with a device and want to help me debug, you're more than welcome to open a new issue and we can try to figure out a fix.