ethz-asl/ethzasl_xsens_driver #55: could not find proper MT device.

wojiaojiao, 2017-08-27:
rosrun xsens_driver mtdevice.py 
is ok!
but
rosrun xsens_driver mtnode.py 
is show erro:could not find proper MT device.
why?

fcolas, 2017-09-19:
It's rather unexpected.
Could it be that you have a parameter set somehow that overrides the default for the ROS node.
Could you try again and paste the output of the following commands:
```
rosrun xsens_driver mtdevice.py -i
rosparam list
rosrun xsens_driver mtnode.py
```
Also does this behaviour happen using the launch file:
```
roslaunch xsens_driver xsens_driver.launch
```

Chambana, 2017-11-22:
I have the same issue with my MTI-G IMU:

sometimes I get this error:
> rosrun xsens_driver mtdevice.py -i
> MTException: Timeout: waiting for message

sometimes I get good output:

> $ rosrun xsens_driver mtdevice.py -i
> Device: /dev/ttyUSB0 at 115200 Bd:
>   device ID:  0x00501266
>   product code: 'MTi-G-28A53G35'
>   firmware revision: (2, 5, 1)
>   baudrate: 2
>   error mode:  0x0001
>   option flags:  message unsupported by your device.
>   location ID:  0x0000
>   transmit delay:  message unsupported by your device.
>   synchronization settings:  message unsupported by your device.
>   general configuration: {   'Master device ID': 5247590,
>     'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
>     'device ID': 5247590,
>     'length': 18,
>     'number of devices': 1,
>     'output-mode': 4,
>     'output-settings': 1,
>     'period': 1152,
>     'skipfactor': 0,
>     'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
>   output configuration (mark IV devices):  message unsupported by your device.
>   string output type:  message unsupported by your device.
>   period: 1152
>   alignment rotation sensor:  message unsupported by your device.
>   alignment rotation local:  message unsupported by your device.
>   output mode:  0x0004
>   extended output mode:  0x0000
>   output settings:  0x00000001
>   GPS coordinates (lat, lon, alt): (36.595211029052734, -121.87519836425781, -19.46299934387207)
>   available scenarios: [   (1, 7, 'general'),
>     (2, 7, 'automotive'),
>     (3, 7, 'aerospace'),
>     (9, 7, 'general_nobaro'),
>     (10, 7, 'aerospace_nobaro')]
>   current scenario ID: 10
>   UTC time: (20000000, 0, 0, 0, 0, 0, 1, 0)
> 

However, this error is consistent whenever I run the ros node:

> $ rosrun xsens_driver mtnode.py
> [INFO] [1511377305.508657]: Found parameter: ~device, value: /dev/ttyUSB0
> [INFO] [1511377305.511945]: Found parameter: ~baudrate, value: 115200
> [INFO] [1511377305.515061]: Found parameter: ~timeout, value: 0.002
> [INFO] [1511377305.515626]: MT node interface: /dev/ttyUSB0 at 115200 bd.
> Traceback (most recent call last):
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 771, in <module>
>     main()
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 766, in main
>     driver = XSensDriver()
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 59, in __init__
>     self.mt = mtdevice.MTDevice(device, baudrate, timeout)
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 43, in __init__
>     self.auto_config_legacy()
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 622, in auto_config_legacy
>     self.GetConfiguration()
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 347, in GetConfiguration
>     self._ensure_config_state()
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 180, in _ensure_config_state
>     self.GoToConfig()
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 208, in GoToConfig
>     self.write_ack(MID.GoToConfig)
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 165, in write_ack
>     mid_ack, data_ack = self.read_msg()
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 132, in read_msg
>     if ord(self.waitfor()) != 0xFA:
>   File "/home/drew/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 88, in waitfor
>     raise MTTimeoutException("waiting for message")
> mtdef.MTTimeoutException: Timeout: waiting for message
> 

Strangely, running the python script shows good output from the MTI-G IMU:

> $ rosrun xsens_driver mtdevice.py 
> {'Sample': 0, 'Orient': {'frame': 'ENU', 'quaternion': (0.30378180742263794, -0.0016711449716240168, 0.0034434483386576176, -0.9527339339256287)}}
> {'Sample': 1, 'Orient': {'frame': 'ENU', 'quaternion': (0.30255454778671265, -0.00152109086047858, 0.003533814335241914, -0.9531242847442627)}}
> {'Sample': 2, 'Orient': {'frame': 'ENU', 'quaternion': (0.3026276230812073, -0.001434833975508809, 0.003639130387455225, -0.953100860118866)}}
> ....

Of note, I have confirmed that the device was added to the dialout group

> ls -l /dev/ttyUSB0
> crw-rw---- 1 root dialout 188, 0 Nov 22 11:02 /dev/ttyUSB0

and I reset the the MIT-G to factory defaults using the -r flag.  No change. 

Also, I am running master 59d3a58eeda5b538c5cbbd1865b230f36e3a1425

fcolas, 2017-11-28:
I'm not sure what's happening here but it might just be some timing issues. Did you try to increase the `timeout` parameter of the node?

JorgeHormaza, 2017-11-28:
I got the same error. However, I'm not sure about the parameters if I have to change something or not. (I mean if this code is able to recognise all parameters from the dongle) because I'm using:
roslaunch xsens_driver. launch 
I get: 
... logging to /home/laura/.ros/log/70dee1ba-d460-11e7-9763-ace0106a1abd/roslaunch-laura-X453MA-4162.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://laura-X453MA:35887/
SUMMARY
========
PARAMETERS
 * /rosdistro: lunar
 * /rosversion: 1.13.2
 * /xsens_driver/baudrate: 0
 * /xsens_driver/device: auto
 * /xsens_driver/frame_id: /imu
 * /xsens_driver/frame_local: ENU
 * /xsens_driver/no_rotation_duration: 0
 * /xsens_driver/timeout: 0.002
NODES
  /
    xsens_driver (xsens_driver/mtnode.py)
ROS_MASTER_URI=http://localhost:11311
process[xsens_driver-1]: started with pid [4179]
[INFO] [1511893280.446581]: Found parameter: ~device, value: auto
[INFO] [1511893280.451477]: Found parameter: ~baudrate, value: 0
[INFO] [1511893280.455882]: Found parameter: ~timeout, value: 0.002
[ERROR] [1511893280.745796]: Fatal: could not find proper MT device.
[xsens_driver-1] process has finished cleanly
log file: /home/laura/.ros/log/70dee1ba-d460-11e7-9763-ace0106a1abd/xsens_driver-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
--------------------------------------------------------------------------------------------------------
also if I use: 
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver$ rosrun xsens_driver mtdevice.py -i
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver$ rosparam list
/rosdistro
/roslaunch/uris/host_laura_x453ma__41167
/roslaunch/uris/host_laura_x453ma__45429
/rosversion
/run_id
/xsens_driver/baudrate
/xsens_driver/device
/xsens_driver/frame_id
/xsens_driver/frame_local
/xsens_driver/no_rotation_duration
/xsens_driver/timeout
-----------------------------------------------------------------------------------------------------------
How do you guys resolve this problem?

awesomebytes, 2017-11-29:
Hello, first @JorgeHormaza could you reformat your comment so it's not all in big bold letters? If you want to paste a piece of code/output you can write in between two triple back-ticks "```" as explained [here](https://help.github.com/articles/creating-and-highlighting-code-blocks/#fenced-code-blocks) it will make it easier to read.

Note that I don't have a device to try any of this.

What version of Ubuntu and ROS are you using? What exact MTi device are you using (MTw [seems to need some tweaks](https://github.com/ethz-asl/ethzasl_xsens_driver/issues/51))? Have you tested that the device actually works (i.e.: on Windows with whatever test tool they provide)?

First I would plug the device and check if your computer is recognising it. (Also if your device is battery powered, make sure it has battery).

Do this before pluging the device in:
```bash
dmesg  | tail
```
Then plug it in, and repeat the same command:
```bash
dmesg | tail
```

You should see some kernel output saying a USB/Serial device/Something was plugged in.

If you don't.... that's a bad start. You could also try:

```bash
lsusb
```
Before and after pluging it in, to see if a USB device appeared. If that also doesn't make it, well your computer is not recognising it.

**If your computer actually recognised it** and you know which serial port is using (for example `/dev/ttyUSB0`) you can try modifying the launch file you are using to instead of using `auto` force it to use that serial port.

The [xsens_driver wiki](http://wiki.ros.org/xsens_driver) also says in the end:

> It might be necessary to add the user to the dialout group so that the node can communicate with the device.

Which would be done with `sudo adduser YOURUSERNAME dialout`. Note that you need to log out and log in (or reboot) after the command for this to take effect. And also unplug the device and plug the device again.

And as an extra test you could also do `sudo chmod +rwa /dev/ttyUSB0` to give all kinds of permissions to the port to be read & write.

And then finally, I can only wish you good luck and remind you that you can try to use the `mtdevice.py --reset` to reset the device in case it was some previous incompatible configuration and `mtdevice.py --inspect` to see what is going on. But all this depends on your computer giving you a serial device to be able to "talk" to it.

JorgeHormaza, 2017-12-01:
Hi @awesomebytes  Thank you so much for answer my message, I followed all the steps but another error is generated. I get:

´´´ ... logging to /home/laura/.ros/log/916d304c-d6dd-11e7-ad6e-ace0106a1abd/roslaunch-laura-X453MA-8285.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laura-X453MA:39501/

SUMMARY
========

PARAMETERS
 * /rosdistro: lunar
 * /rosversion: 1.13.2
 * /xsens_driver/baudrate: 961200
 * /xsens_driver/device: /dev/ttyUSB1
 * /xsens_driver/frame_id: /imu
 * /xsens_driver/frame_local: ENU
 * /xsens_driver/no_rotation_duration: 0
 * /xsens_driver/timeout: 0.5

NODES
  /
    xsens_driver (xsens_driver/mtnode.py)

ROS_MASTER_URI=http://localhost:11311

process[xsens_driver-1]: started with pid [8302]
[INFO] [1512163374.491012]: Found parameter: ~device, value: /dev/ttyUSB1
[INFO] [1512163374.496107]: Found parameter: ~baudrate, value: 961200
[INFO] [1512163374.499837]: Found parameter: ~timeout, value: 0.5
[INFO] [1512163374.500456]: MT node interface: /dev/ttyUSB1 at 961200 bd.
Traceback (most recent call last):
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 772, in <module>
    main()
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 767, in main
    driver = XSensDriver()
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 59, in __init__
    self.mt = mtdevice.MTDevice(device, baudrate, timeout)
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 43, in __init__
    self.auto_config_legacy()
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 622, in auto_config_legacy
    self.GetConfiguration()
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 347, in GetConfiguration
    self._ensure_config_state()
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 180, in _ensure_config_state
    self.GoToConfig()
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 208, in GoToConfig
    self.write_ack(MID.GoToConfig)
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 165, in write_ack
    mid_ack, data_ack = self.read_msg()
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 132, in read_msg
    if ord(self.waitfor()) != 0xFA:
  File "/home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 88, in waitfor
    raise MTTimeoutException("waiting for message")
mtdef.MTTimeoutException: Timeout: waiting for message
[xsens_driver-1] process has died [pid 8302, exit code 1, cmd /home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py __name:=xsens_driver __log:=/home/laura/.ros/log/916d304c-d6dd-11e7-ad6e-ace0106a1abd/xsens_driver-1.log].
log file: /home/laura/.ros/log/916d304c-d6dd-11e7-ad6e-ace0106a1abd/xsens_driver-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
laura@laura-X453MA:~$ cd catkin_ws/src/ethzasl_xsens_driver/
.git/   launch/ nodes/  
laura@laura-X453MA:~$ cd catkin_ws/src/ethzasl_xsens_driver/nodes/
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ python mtdevice.py
MTException: unable to open /dev/ttyUSB0
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ python mtdevice.py --reset
MTException: unable to open /dev/ttyUSB0
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ python mtdevice.py -d /dev/ttyUSB1 -- reset
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ python mtdevice.py -d /dev/ttyUSB1 --inspect
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ python mtdevice.py -d /dev/ttyUSB1 --reset
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ python mtdevice.py -d /dev/ttyUSB1
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ ./mtdevice.py mt-r
MTException: unable to open /dev/ttyUSB0
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ ./mtdevice.py -d /dev/ttyUSB1 -r
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ ./mtdevice.py -d /dev/ttyUSB1 -b 0
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ ./mtdevice.py -d /dev/ttyUSB1 -b 0
MTException: Timeout: waiting for message
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ python mtdevice.py -d /dev/ttyUSB0 -b 921600 -i^C
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ ^C
laura@laura-X453MA:~/catkin_ws/src/ethzasl_xsens_driver/nodes$ 

´´´

this is the answer when I used this command

´´´roslaunch xsens_driver xsens_driver.launch ´´´

How you can see the port is recognized, but the timeout send:  "wait for message" and it generates this error that is unexpected and I don't know how solution this:

´´´ mtdef.MTTimeoutException: Timeout: waiting for message
[xsens_driver-1] process has died [pid 8302, exit code 1, cmd /home/laura/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py __name:=xsens_driver __log:=/home/laura/.ros/log/916d304c-d6dd-11e7-ad6e-ace0106a1abd/xsens_driver-1.log].
log file: /home/laura/.ros/log/916d304c-d6dd-11e7-ad6e-ace0106a1abd/xsens_driver-1*.log ´´´

fcolas, 2017-12-08:
For the node, you have set the parameter `baudrate` to 961200. Did you try to configure your device with such baudrate?
If so, I mentioned on the wiki that some devices might not actually support all the baudrates they are supposed to: I had to use the emergency reset procedure (not just reset to factory default take a look at your manual).

fcolas, 2018-03-06:
Any news on that?

fcolas, 2018-04-04:
@wojiaojiao @Chambana @JorgeHormaza I'm going to close the issue if you don't have it any more.

JorgeHormaza, 2018-04-04:
So sorry Mate, I had to quite with this project for personal reasons. However,  Thank you so much for your help.

Cordialmente;
Jorge Andres Hormaza Parra
Estudiante Ing. Biomédica

jorge.hormaza@mail.escuelaing.edu.co

Cel 311 2348385

Universidad del Rosario - Escuela Colombiana de Ingeniería Julio Garavito

qingwenZhang, 2019-07-26:
> I'm not sure what's happening here but it might just be some timing issues. Did you try to increase the `timeout` parameter of the node?

You are right! Just increase 'timeout' in xsens_driver.launch!