ethz-asl/ethzasl_xsens_driver #58: Driver output shows strange behavior

DLabo, 2017-10-27:
Hello,

I am using the Xsense MTi-30 both with ROS kinetic and Ubuntu 16.04 and ROS indigo and 14.04. The data obtained looked correct but when I visualized them with rqt-plot I found a strange behavior in the output (screnschot attached). Has someone already dealt with this problem?

Thanks
[image]

Firmware: 1.7.2 build 42 rev 64903   Hardware: 2.1.0

fcolas, 2017-11-28:
This looks strange and I don't observe the same thing. Does it happen in both indigo/14.04 and kinetic/16.04?

Can you show the output of:
```rosrun xsens_driver mtdevice.py -i```
Also of:
```rostopic hz /imu/data```

One explanation I could think of is that for some reason the data is buffered and published in bursts. Can you try to show when the data is published by editing the properties of the graph and increasing the size of the circle to 5 or 10, for instance (the checkbox besides the save icon in `rqt_plot`).
On my MTi-30 I get a smoother graph (see below).
[image]

hadiTab, 2018-01-30:
Are there any solutions for this yet?

I'm having the same issue with an MTI-3.
The problem is in the time stamps for the ros messages. Each of those near vertical parts in the graph is 5 consecutive messages that are being stamped too close to each other. The sensor data looks right in MT Manager though.

The frequency of the published messages is at 100 (+/- 2 hz).

running ```rosrun xsens_driver mtdevice.py -i``` returns:
```
MTException: could not find message.
```

DLabo, 2018-01-30:
I could not investigate more this problem but I found out that, in my case, it was related with the cable I was using (CA-USB2-MTI). By connecting my MTi-30 with a  CA-USB-MTI I have a regular behavior.

fcolas, 2018-03-06:
That's an interesting finding; many thanks @DLabo 
I'm closing it seems it doesn't seem to be related to the driver.

closed 2018-03-06 by fcolas