ethz-asl/ethzasl_xsens_driver #61: 5 th generation

0xd1ma, 2017-12-12:
Hello everyone.

For autonomous vehicle project, I am planning to by MTi-30 AHRS.

On http://wiki.ros.org/xsens_driver says: "This package provides a driver for the third and fourth generation..."

Do this ROS driver supports 5 th generation of Xsens sensors?

Thank you.

fcolas, 2017-12-13:
I'm not aware that Xsens updated their sensors to a fifth generation and, to my knowledge, MTi-30 AHRS is a name that fits the MTi-10 series introduced as the fourth generation.
Can you provide a link for that?
In any case, as long as it complies to the same protocol with the same message type on a serial or USB interface, it should be supported.

0xd1ma, 2017-12-14:
first page and Introduction
[MTi_usermanual.pdf](https://github.com/ethz-asl/ethzasl_xsens_driver/files/1559083/MTi_usermanual.pdf)

fcolas, 2018-03-06:
Sorry for the delay. I was finally able to find updated documentation and it seems that they only added a few things but most of the protocol is similar. I expect most things will work and but the new feature are not yet configurable. I'm going to work on that.

fcolas, 2018-04-04:
@0xd1ma Did you buy the device? If so, could you test the `support_gen_5` branch?

0xd1ma, 2018-04-21:
@fcolas the sensor will arrive in mid-May. I will test

0xd1ma, 2018-06-19:
Finally, the sensor arrived and I performed some tests.
Brunch works.
but
from time to time the node stops working and issues the following:
`SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /xsens_driver/baudrate: 115200
 * /xsens_driver/device: /dev/mti30
 * /xsens_driver/frame_id: imu
 * /xsens_driver/frame_local: ENU
 * /xsens_driver/no_rotation_duration: 0
 * /xsens_driver/timeout: 0.002

NODES
  /
    xsens_driver (xsens_driver/mtnode.py)

ROS_MASTER_URI=http://localhost:11311

process[xsens_driver-1]: started with pid [4780]
[INFO] [1529336639.413169]: Found parameter: ~device, value: /dev/mti30
[INFO] [1529336639.416146]: Found parameter: ~baudrate, value: 115200
[INFO] [1529336639.419158]: Found parameter: ~timeout, value: 0.002
[INFO] [1529336639.419691]: MT node interface: /dev/mti30 at 115200 bd.
[INFO] [1529336639.478598]: Found parameter: ~no_rotation_duration, value: 0
[INFO] [1529336639.479631]: Found parameter: ~frame_id, value: imu
[INFO] [1529336639.480527]: Found parameter: ~frame_local, value: ENU
Traceback (most recent call last):
  File "/home/scar/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 780, in <module>
    main()
  File "/home/scar/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 776, in main
    driver.spin()
  File "/home/scar/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 133, in spin
    self.spin_once()
  File "/home/scar/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 692, in spin_once
    data = self.mt.read_measurement()
  File "/home/scar/git/ethzasl_xsens_driver/nodes/mtdevice.py", line 683, in read_measurement
    return self.parse_MTData2(data)
  File "/home/scar/git/ethzasl_xsens_driver/nodes/mtdevice.py", line 1014, in parse_MTData2
    raise MTException("couldn't parse MTData2 message.")
mtdef.MTException: couldn't parse MTData2 message.
[xsens_driver-1] process has died [pid 4780, exit code 1, cmd /home/scar/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode.py __name:=xsens_driver __log:=/home/scar/.ros/log/51d8dcec-730e-11e8-a560-d85d4c888192/xsens_driver-1.log].
log file: /home/scar/.ros/log/51d8dcec-730e-11e8-a560-d85d4c888192/xsens_driver-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

There are also no reports of speed changes, similarly https://github.com/ethz-asl/ethzasl_xsens_driver/issues/73 
`---
header: 
  seq: 2418
  stamp: 
    secs: 1529338273
    nsecs: 578365087
  frame_id: "imu"
twist: 
  linear: 
    x: 0.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0288859009743
    y: -0.0141263036057
    z: -0.00568330287933
---
`

[mtdevice.txt](https://github.com/ethz-asl/ethzasl_xsens_driver/files/2115243/mtdevice.txt)
[xsens.txt](https://github.com/ethz-asl/ethzasl_xsens_driver/files/2115244/xsens.txt)

fcolas, 2018-06-20:
Many thanks for the information.
The issue you have seems to come from parsing the message from the IMU. Is it easy to reproduce?
Does it happen without the node (there's no reason it wouldn't):

    rosrun mtdevice.py

If that's the case, you could try with the verbose flag and paste the (end of the) output of:

    rosrun mtdevice.py -v

This would allow us to see the content of the message that poses problem.

fcolas, 2018-06-28:
@0xd1ma No news?

Adixy, 2018-09-19:
I edited mtdevice.py script and just put "pass" instead the "raise Exception(" couldn't parse MTData2 message.") thing.

It reads all the sensor data correctly in real time. Probably some typo in the script somewhere that triggers try/except

fcolas, 2018-10-26:
@Adixy Interesting.
As you can see from the contents of the last message that the driver cannot parse, there is a GPS_SOL message of length 52 (0x34), a SampleTimeFine of length 4, and a PacketCounter of length 2. The latter two are obviously ok but, according to the documentation, the GPS_SOL message should have a length of 49.
Note that this message type got deprecated in favour of Position or GNSS types of messages. You can try to switch to one of those.

Anyway, if we wanted to solve this specific issue, we would have to guess what the 52 bytes mean. The easiest explanation is that, somewhere, a 1-byte field has been upgraded to 4 bytes. Looking inside this message is looks ok up to the end. So it might just be that the last reserved byte is actually not a byte but a float (1.28e-40?), an int (91938?) or some other 4-byte flag.
I'll try to hack in a fix. But in the end, it's probably not the same issue since OP cannot have such message.

fcolas, 2018-10-26:
@Adixy Can you please test the [gps_sol_workaround](https://github.com/ethz-asl/ethzasl_xsens_driver/tree/gps_sol_workaround) branch?

fcolas, 2019-08-25:
@Adixy Ping?