ethz-asl/ethzasl_xsens_driver #63: Configuration parameteres mti-3 1-series

matteopantano, 2018-02-28:
Hey guys,

I am struggling with the configuration of a mti-3. I would like to know if I can tell to the xsens to activate representative motion and ICC otherwise the reading is not consistent. Furthermore I would like to deactivate AHS.

Thanks

fcolas, 2018-03-06:
These are new features I didn't know existed. It seems to have been introduced sometimes last year, maybe along with mk5 devices.
I'll try to take a quick look but if not too many things changed, it could be relatively quick to add.

fcolas, 2018-03-14:
I've started to code the features for this support, but I don't have access to a fifth generation device to test it.
Could you take a look at the `support_gen_5` branch and test it, please?
The new features are mainly the `--gnss-platform`, `--option_flags`, and `--icc-command` command line arguments to `mtdevice.py`. Obviously you cannot really test the first one, but you should be able to play around with the other two, especially to activate representative motion and ICC and to toggle AHS.

fcolas, 2018-04-04:
@matteopantano Any news? It'd be great if you could help me test the functionality you requested.

fcolas, 2019-08-25:
@matteopantano Ping?

fcolas, 2019-09-06:
I've merged the branch.

closed 2019-09-06 by fcolas