ethz-asl/ethzasl_xsens_driver #66: Can not use fix topic on MTi-G

hecperleo, 2018-03-23:
Hi, 

I am trying to use MTi-G with GPS on ROS. 

Most of the time I get this output 
```
rosrun xsens_driver mtdevice.py -i
MTException: Timeout: waiting for message

```

Sometimes I get this output
```
rosrun xsens_driver mtdevice.py -i
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x01500618
  product code: 'MTi-G-28A53G35'
  firmware revision: (2, 6, 1)
  baudrate: 2
  error mode:  MTException: Timeout: waiting for message

```

If I run this command I get this output
```
roslaunch xsens_driver xsens_driver.launch
[WARN] [1521809140.589618]: Cannot find value for parameter: ~device, assigning default: auto
[WARN] [1521809140.592510]: Cannot find value for parameter: ~baudrate, assigning default: 0
[WARN] [1521809140.595244]: Cannot find value for parameter: ~timeout, assigning default: 0.002
[INFO] [1521809140.672131]: Detected MT device on port /dev/ttyUSB0 @ 115200 bps
[INFO] [1521809140.672670]: MT node interface: /dev/ttyUSB0 at 115200 bd.
[INFO] [1521809140.724260]: Found parameter: ~frame_id, value: /imu
[INFO] [1521809140.727137]: Found parameter: ~frame_local, value: ENU
```

By the way, I am able to see 
```
rostopic list
/imu/correction_angle
/imu/correction_angle_pitch
/imu/correction_angle_roll
/imu/data
/imu/mag
/imu/view_data
/imu_data_str
/rosout
/rosout_agg
/velocity

```

But I have GPS connected also so I want to see longitude, latitude, and altitude. Any solution?

Thanks.

fcolas, 2018-03-23:
Getting those topics should be pretty easy once the `Timeout` exception is solved: it should just be a change in configuration.

For this exception, adding the verbose flag might help us get more information; can you try `rosrun xsens_driver mtdevice.py -i -v`?

Cables might have an impact on the communication (see #58).
The simplest temporary fix could be to try increasing the timeout in [`mtdevice.py`](https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtdevice.py#L23) to `0.01` or even higher. Please report whether it solves your issue or not, and what value was necessary (which worked reliably and which didn't).

hecperleo, 2018-03-26:
To get more information I did this: 

```
rosrun xsens_driver mtdevice.py -i -v
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [01 50 06 18 04 80 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 01 50 06 18 00 36 00 06 00 00 00 01 07 0A 02 06 01 36 02 02]
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  MT: Write message id 0x00 (ReqDID) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x01 (DeviceID) with 4 data bytes: [01 50 06 18]
0x01500618
  product code:  MT: Write message id 0x1C (ReqProductCode) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x1D (ProductCode) with 14 data bytes: [4D 54 69 2D 47 2D 32 38 41 35 33 47 33 35]
'MTi-G-28A53G35'
  firmware revision:  MT: Write message id 0x12 (ReqFWRev) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x13 (FirmwareRev) with 3 data bytes: [02 06 01]
(2, 6, 1)
  baudrate:  MT: Write message id 0x18 (SetBaudrate) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x19 (SetBaudrateAck) with 1 data bytes: [02]
2
  error mode:  MT: Write message id 0xDA (SetErrorMode) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0xDB (SetErrorModeAck) with 2 data bytes: [00 01]
[... 190 lines cut ...]
  firmware revision:  MT: Write message id 0x12 (ReqFWRev) with 0 data bytes: []
MT: Got message id 0x13 (FirmwareRev) with 3 data bytes: [02 06 01]
(2, 6, 1)
  baudrate:  MT: Write message id 0x18 (SetBaudrate) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
```

If I increase the timeout a lot, for example 0.1 in mtdevice.py I get this output without the -v flag: 

```
rosrun xsens_driver mtdevice.py -i
MTException: unable to open /dev/ttyUSB0
```

I guess it is too much. By the way, I gave permissions running this:
`
sudo adduser YOURUSERNAME dialout`

```

Thanks for your quick response.

fcolas, 2018-04-04:
The last point (`unable to open .dev.ttyUSB0`) is a different issue, you might accidentally have tried running `mtdevice.py` while the device was already opened (for instance another terminal with either the node or `mtdevice.py` running).
I'm not sure I understand what happens when you set the timeout to `0.01`. Do you really get only the 6 lines you pasted? If you try without the `-v` flag, does it work properly?

hecperleo, 2018-04-04:
Retrying `rosrun xsens_driver mtdevice.py -i -v` without other terminals open, I get this output:

```
rosrun xsens_driver mtdevice.py -i -v

MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x32 (MTData) with 54 data bytes: [3D 61 84 99 BD 25 59AD 41 1D 13 ED 3B 1B 52 60 BA C1 12 34 BB 7F 21 3F BE 3A 38 CC 3E 49 C1 0CBE A3 FC DA 3E 8B AD 87 BB 53 02 CB 3B F3 D3 BA BF 76 47 E0 4B 33]
ack (0x31) expected, got 0x32 instead
MT: Got message id 0x32 (MTData) with 54 data bytes: [3D A6 C6 84 BD 57 6F7E 41 1D 09 6D 3C 08 2A 01 BC 07 80 3C 3A 60 94 CB BE 3A 15 78 3E 4A 34 50BE A4 55 6A 3E 8B AE 07 BB 55 29 D1 3B F1 A4 08 BF 76 47 D5 4B 34]
ack (0x31) expected, got 0x32 instead
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
waiting for 119 bytes, got 31 so far: [01 50 06 18 04 80 00 00 00 10 00 0000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
waiting for 119 bytes, got 31 so far: [01 50 06 18 04 80 00 00 00 10 00 0000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
MT: Got message id 0x0D (Configuration) with 118 data bytes: [01 50 06 18 04 80 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 01 50 06 18 00 36 00 06 00 00 00 01 07 0A 02 06 01 36 02 02]
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  MT: Write message id 0x00 (ReqDID) with 0 data bytes: []
MT: Got message id 0x01 (DeviceID) with 4 data bytes: [01 50 06 18]
0x01500618
  product code:  MT: Write message id 0x1C (ReqProductCode) with 0 data bytes: []
MT: Got message id 0x1D (ProductCode) with 14 data bytes: [4D 54 69 2D 47 2D 32 38 41 35 33 47 33 35]
'MTi-G-28A53G35'
  firmware revision:  MT: Write message id 0x12 (ReqFWRev) with 0 data bytes: []
MT: Got message id 0x13 (FirmwareRev) with 3 data bytes: [02 06 01]
(2, 6, 1)
  baudrate:  MT: Write message id 0x18 (SetBaudrate) with 0 data bytes: []
MT: Got message id 0x19 (SetBaudrateAck) with 1 data bytes: [02]
2
  error mode:  MT: Write message id 0xDA (SetErrorMode) with 0 data bytes:[]
MT: Got message id 0xDB (SetErrorModeAck) with 2 data bytes: [00 01]
0x0001
  option flags:  MT: Write message id 0x48 (SetOptionFlags) with 0 data bytes: []
MT: Got message id 0x42 (Error) with 1 data bytes: [04]
message unsupported by your device.
  location ID:  MT: Write message id 0x84 (SetLocationID) with 0 data bytes: []
MT: Got message id 0x85 (SetLocationIDAck) with 2 data bytes: [00 00]
0x0000
  transmit delay:  MT: Write message id 0xDC (SetTransmitDelay) with 0 data bytes: []
MT: Got message id 0x42 (Error) with 1 data bytes: [04]
message unsupported by your device.
  synchronization settings:  MT: Write message id 0x2C (SetSyncSettings) with 0 data bytes: []
MT: Got message id 0x42 (Error) with 1 data bytes: [04]
message unsupported by your device.
  general configuration:  MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
waiting for 119 bytes, got 49 so far: [01 50 06 18 04 80 00 00 00 10 00 0000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0000 00 00 00 00 00 00 00 00 00 00 00]
waiting for 119 bytes, got 49 so far: [01 50 06 18 04 80 00 00 00 10 00 0000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0000 00 00 00 00 00 00 00 00 00 00 00]
MT: Got message id 0x0D (Configuration) with 118 data bytes: [01 50 06 18 04 80 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 01 50 06 18 00 36 00 06 00 00 00 01 07 0A 02 06 01 36 02 02]
{   'Master device ID': 22021656,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 22021656,
    'length': 54,
    'number of devices': 1,
    'output-mode': 6,
    'output-settings': 1,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  MT: Write message id 0xC0 (SetOutputConfiguration) with 0 data bytes: []
MT: Got message id 0x42 (Error) with 1 data bytes: [04]
message unsupported by your device.
  string output type:  MT: Write message id 0x8E (SetStringOutputType) with 0 data bytes: []
MT: Got message id 0x42 (Error) with 1 data bytes: [04]
message unsupported by your device.
  period:  MT: Write message id 0x04 (SetPeriod) with 0 data bytes: []
MT: Got message id 0x05 (SetPeriodAck) with 2 data bytes: [04 80]
1152
  alignment rotation sensor:  MT: Write message id 0xEC (SetAlignmentRotation) with 1 data bytes: [00]
MT: Got message id 0x42 (Error) with 1 data bytes: [04]
[... 6 lines cut ...]
0x0006
  extended output mode:  MT: Write message id 0x86 (SetExtOutputMode) with0 data bytes: []
MT: Got message id 0x87 (SetExtOutputModeAck) with 2 data bytes: [00 00]
0x0000
  output settings:  MT: Write message id 0xD2 (SetOutputSettings) with 0 data bytes: []
MT: Got message id 0xD3 (SetOutputSettingsAck) with 4 data bytes: [00 00 00 01]
0x00000001
  GPS coordinates (lat, lon, alt):  MT: Write message id 0x6E (SetLatLonAlt) with 0 data bytes: []
[... cut ...]
[   (1, 7, 'general'),
    (2, 7, 'automotive'),
    (3, 7, 'aerospace'),
    (9, 7, 'general_nobaro'),
    (10, 7, 'aerospace_nobaro')]
  current scenario ID:  MT: Write message id 0x64 (SetCurrentScenario) with 0 data bytes: []
MT: Got message id 0x65 (SetCurrentScenarioAck) with 2 data bytes: [00 0A]
10
  UTC time:  MT: Write message id 0x60 (SetUTCTime) with 0 data bytes: []
MT: Got message id 0x61 (UTCTime) with 12 data bytes: [1F 97 28 80 00 00 00 00 00 03 0C 00]
(530000000, 0, 0, 0, 0, 3, 12, 0)
```

fcolas, 2018-04-04:
Ok, so if you don't get any more timeout error with this setting, you should be able to configure your device to output GPS information (see legacy configuration in the documentation).
I'll try to add a command line parameter for a longer timeout.

fcolas, 2018-04-04:
@hecperleo I've pushed the `timeout_param` branch adding a command line argument (`-t` or `--timeout`) to `mtdevice.py` so as to pass a different timeout value than the default (`0.002`).
Could you test it?

hecperleo, 2018-04-04:
Thanks! I am not used to python. I saw legacy configuration but I was not able to do it right. I wrote 
```
rosrun xsens_driver mtdevice.py -m --output-mode=pv 
Unknown output-mode specifier: '-'
```

fcolas, 2018-04-04:
Well, it's more about command line arguments (in the standard GNU way) than about Python.
Your current configuration can be obtained with:
```
rosrun xsens_driver mtdevice.py -l --output-mode=co --output-settings=tqMAG
```

If you want to additionally get position and velocity from the IMU you can try:
```
rosrun xsens_driver mtdevice.py -l --output-mode=copv --output-settings=tqMAG
```

hecperleo, 2018-04-04:
Thanks you! 

Now I can see and use /fix and all IMU topics.

closed 2018-04-04 by hecperleo

hecperleo, 2018-04-04:
Yes, it solves my timeout issue. I forgot to confirm it.

fcolas, 2018-04-04:
Many thanks!