ethz-asl/ethzasl_xsens_driver #67: Error with mtdevice.py -i since firmware upgrade to 1.8.2

moooeeeep, 2018-03-28:
Since we upgraded the firmware on our devices to version 1.8.2 we observe errors with the `mtdevice.py -i` script. Apparantly the response received is longer than expected, and thus it fails to parse given the pattern.

Here is the output for example:

	$ rosrun xsens_driver mtdevice.py -i
	Device: /dev/ttyUSB0 at 115200 Bd:
	  device ID:  0x077003D4
	  product code: 'MTi-G-700-2A5G4'
	  firmware revision: (1, 8, 2)
	  baudrate: 2
	  error mode:  message unsupported by your device.
	  option flags:  0x00000010
	  location ID:  0x0000
	  transmit delay:  message unsupported by your device.
	  synchronization settings:  [(0x09, 0x01, 0x01, 0x00, 0x0000, 0x0000, 0x0000, 0x00FA)]
	  general configuration: {   'Master device ID': 124781524,
	    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
	    'device ID': 124781524,
	    'length': 0,
	    'number of devices': 1,
	    'output-mode': 0,
	    'output-settings': 1,
	    'period': 1152,
	    'skipfactor': 0,
	    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
	  output configuration (mark IV devices):  [(0x1020, 65535), (0x1060, 65535), (0x2013, 400), (0x4023, 400), (0x8023, 400), (0x5043, 400), (0xD013, 400)]
	  string output type: 0
	  period: 1152
	  alignment rotation sensor: 
	Traceback (most recent call last):
	  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 1704, in <module>
	    main()
	  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 1430, in main
	    inspect(mt, device, baudrate)
	  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 1530, in inspect
	    parameter=0)
	  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 1508, in try_message
	    pprint.pprint(f(*args, **kwargs), indent=4)
	  File "/opt/ros/kinetic/lib/xsens_driver/mtdevice.py", line 424, in GetAlignmentRotation
	    q0, q1, q2, q3 = struct.unpack('!ffff', data)
	struct.error: unpack requires a string argument of length 16

The actual data received is this: `bytearray(b'\x00?\x80\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')` (length: 17).

When I slice off a byte at the front or at the back of the buffer, the script continues without any other obvious errors raised. The device produces data seemingly normal (as long as you don't configure it to produce fixed-point GPS coordinates, I recall). We observed occasional crashes recently, for which we didn't find the cause so far.

Now I wonder, did we break the device (with the firmware update?) or is the driver incompatible? Can anyone confirm the problem or help resolve the issue?

fcolas, 2018-04-04:
I just pushed a new branch `fix_get_alignment_rotation` which should fix the issue, please check it out.
I believe it was just a bug in my code (I didn't read the specs correctly) that didn't appear before (most devices seem to not answer this message). Do you remember if you ever got a value for this `alignment rotation sensor/local`?

tbarchyn, 2018-04-11:
Possibly of help: I had the same error with a new MTi-3, which I presume is close to the latest firmware. The `fix_get_alignment_rotation` branch repairs it for me.

The output from `mtdevice.py -i` is working now:
```
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x03881F5C
  product code: 'MTi-3-8A7G6'
  firmware revision: (1, 2, 1)
  baudrate: 2
  error mode:  timeout: might be unsupported by your device.
  option flags:  0x00000000
  location ID:  0x0000
  transmit delay:  timeout: might be unsupported by your device.
  synchronization settings:  []
  general configuration: {   'Master device ID': 59252572,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 59252572,
    'length': 888,
    'number of devices': 1,
    'output-mode': 0,
    'output-settings': 0,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  [(0x1020, 65535), (0x1060, 65535), (0x2030, 20), (0xE020, 65535)]
  string output type:  timeout: might be unsupported by your device.
  period: 1152
  alignment rotation sensor: (0, (1.0, 0.0, 0.0, 0.0))
  alignment rotation local: (1, (0.0, -0.7071067690849304, -0.7071067690849304, -0.0))
  output mode:  0x0000
  extended output mode:  timeout: might be unsupported by your device.
  output settings:  0x00000000
  GPS coordinates (lat, lon, alt): (49.1777, -123.92964, 0.0)
  available scenarios: [   (50, 8, 'general'),
    (51, 8, 'high_mag_dep'),
    (52, 8, 'dynamic'),
    (53, 8, 'north_reference'),
    (54, 8, 'vru_general')]
  current scenario ID: 50
  UTC time: (0, 1970, 1, 1, 0, 0, 0, 0)
```

fcolas, 2018-04-11:
Great, thanks for your feedback. I'll merge it in, then.

fcolas, 2018-04-11:
Fixed in #71

closed 2018-04-11 by fcolas