ethz-asl/ethzasl_xsens_driver #73: No linear velocity in Velocity message

nflsalex, 2018-05-24:
I am using MTi-G-710. It outputs Velocity messages but all the linear velocity message have 0 values. Can someone help me with this?

fcolas, 2018-05-24:
Did you make sure to configure your device so as to output the linear velocity?
Can you show us the output of:

    $ rosrun xsens_driver mtdevice.py -i

and a few lines of:
 
    $ rosrun xsens_driver mtdevice.py

willdzeng, 2018-06-08:
I am having the same issue, this is what I got:
```
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x077010E6
  product code: 'MTi-G-710-2A5G4'
  firmware revision: (1, 4, 12)
  baudrate: 2
  error mode:  message unsupported by your device.
  option flags:  0x00000000
  location ID:  0x0000
  transmit delay:  message unsupported by your device.
  synchronization settings:  [(0x09, 0x01, 0x01, 0x00, 0x0000, 0x0000, 0x0000, 0x00FA)]
  general configuration: {   'Master device ID': 124784870,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 124784870,
    'length': 0,
    'number of devices': 1,
    'output-mode': 6,
    'output-settings': 1,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  [(0x1020, 65535), (0x1060, 65535), (0x1010, 65535), (0x2030, 100), (0x4020, 100), (0x4010, 100), (0x8020, 100), (0x8030, 100), (0xE020, 65535), (0x5042, 100), (0x5022, 100), (0xD012, 100), (0x7010, 4), (0x8840, 4), (0x1030, 65535)]
  string output type: 0
  period: 1152
  alignment rotation sensor: 
Traceback (most recent call last):
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1927, in <module>
    main()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1569, in main
    inspect(mt, device, baudrate)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1688, in inspect
    "alignment rotation sensor:", mt.GetAlignmentRotation, parameter=0)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1663, in try_message
    pprint.pprint(f(*args, **kwargs), indent=4)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 447, in GetAlignmentRotation
    param, q0, q1, q2, q3 = struct.unpack('!Bffff', data)
struct.error: unpack requires a string argument of length 17
```

```
{'Acceleration': {'Delta v.z': 0.09767693281173706, 'Delta v.y': 0.001790186041034758, 'Delta v.x': -0.005405271425843239, 'frame': 'ENU', 'accY': 0.17899639904499054, 'accX': -0.5405529141426086, 'accZ': 9.767691612243652}, 'Timestamp': {'Hour': 21, 'PacketCounter': 61610, 'Month': 6, 'Second': 0, 'Flags': 247, 'Year': 2018, 'ns': 715200000, 'Day': 8, 'Minute': 28, 'SampleTimeFine': 11696063}, 'Orientation Data': {'Yaw': 0.026408430188894272, 'Pitch': 2.852447509765625, 'frame': 'ENU', 'Roll': 0.8532515168190002}, 'Angular Velocity': {'Delta q1': -2.1992852339280944e-07, 'Delta q0': 1.0000001192092896, 'Delta q3': -1.6555204638279974e-05, 'Delta q2': 1.0131808494406869e-06, 'frame': 'ENU', 'gyrZ': -0.003311040811240673, 'gyrX': -4.398570308694616e-05, 'gyrY': 0.00020263616170268506}, 'Status': {'StatusWord': 25166021}}

```

willdzeng, 2018-06-08:
I guess it must be a configuration problem, is there any defualt command to run to configure the device correctly? so that it can send acceleration, velocity and position at the same time?

fcolas, 2018-06-11:
@willdzeng The exception you get during inspection is fixed on master: I suggest updating to use that (I should do a new release, I guess).
Looking at your configuration, I see that you should have velocity, but the velocity (0xD012) and position (0x5042 and 0x5022) messages will be send as fixed point which, as you know (#74) is not supported by this driver.
I suggest converting to floating point and try again.
Note that I'm not sure whether the device sends any position or velocity message before it gets a fix, which can take some time.

willdzeng, 2018-06-11:
Thansk for the reply!
I still got this  error from master, when I run
```
rosrun xsens_driver mtdevice.py -i
```
```
Traceback (most recent call last):
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1902, in <module>
    main()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1547, in main
    inspect(mt, device, baudrate)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1666, in inspect
    parameter=0)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1642, in try_message
    pprint.pprint(f(*args, **kwargs), indent=4)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 428, in GetAlignmentRotation
    param, q0, q1, q2, q3 = struct.unpack('!Bffff', data)
struct.error: unpack requires a string argument of length 17
```

willdzeng, 2018-06-11:
Also after change to floating point, I do get everything now.

fcolas, 2018-06-12:
@willdzeng Great to hear you get everything.
I'd like to fix the inspection issue. Could you please paste the output of:

    rosrun xsens_driver mtdevice.py -v -i

willdzeng, 2018-06-13:
> MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
> MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
> waiting for 119 bytes, got 60 so far: [07 70 10 E6 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
> MT: Got message id 0x0D (Configuration) with 118 data bytes: [07 70 10 E6 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 07 70 10 E6 00 00 00 36 00 00 00 03 00 04 01 04 0C 37 04 01]
> Device: /dev/ttyUSB0 at 115200 Bd:
>   device ID:  MT: Write message id 0x00 (ReqDID) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x01 (DeviceID) with 4 data bytes: [07 70 10 E6]
> 0x077010E6
>   product code:  MT: Write message id 0x1C (ReqProductCode) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x1D (ProductCode) with 20 data bytes: [4D 54 69 2D 47 2D 37 31 30 2D 32 41 35 47 34 20 20 20 20 20]
> 'MTi-G-710-2A5G4'
>   firmware revision:  MT: Write message id 0x12 (ReqFWRev) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x13 (FirmwareRev) with 11 data bytes: [01 04 0C 00 00 00 0D 00 00 D5 76]
> (1, 4, 12)
>   baudrate:  MT: Write message id 0x18 (SetBaudrate) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x19 (SetBaudrateAck) with 1 data bytes: [02]
> 2
>   error mode:  MT: Write message id 0xDA (SetErrorMode) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x42 (Error) with 5 data bytes: [04 00 00 00 00]
> message unsupported by your device.
>   option flags:  MT: Write message id 0x48 (SetOptionFlags) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x49 (SetOptionFlagsAck) with 4 data bytes: [00 00 00 00]
> 0x00000000
>   location ID:  MT: Write message id 0x84 (SetLocationID) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x85 (SetLocationIDAck) with 2 data bytes: [00 00]
> 0x0000
>   transmit delay:  MT: Write message id 0xDC (SetTransmitDelay) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x42 (Error) with 5 data bytes: [04 00 00 00 00]
> message unsupported by your device.
>   synchronization settings:  MT: Write message id 0x2C (SetSyncSettings) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x2D (SetSyncSettingsAck) with 12 data bytes: [09 01 01 00 00 00 00 00 00 00 00 FA]
> [(0x09, 0x01, 0x01, 0x00, 0x0000, 0x0000, 0x0000, 0x00FA)]
>   general configuration:  MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
> waiting for 119 bytes, got 60 so far: [07 70 10 E6 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
> MT: Got message id 0x0D (Configuration) with 118 data bytes: [07 70 10 E6 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 07 70 10 E6 00 00 00 36 00 00 00 03 00 04 01 04 0C 37 04 01]
> {   'Master device ID': 124784870,
>     'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
>     'device ID': 124784870,
>     'length': 0,
>     'number of devices': 1,
>     'output-mode': 54,
>     'output-settings': 3,
>     'period': 1152,
>     'skipfactor': 0,
>     'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
>   output configuration (mark IV devices):  MT: Write message id 0xC0 (SetOutputConfiguration) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0xC1 (SetOutputConfigurationAck) with 44 data bytes: [10 20 FF FF 10 60 FF FF 20 10 00 64 40 10 00 64 80 30 00 64 C0 20 00 64 E0 20 FF FF 50 40 00 64 50 20 00 64 D0 10 00 64 70 10 00 04]
> [(0x1020, 65535), (0x1060, 65535), (0x2010, 100), (0x4010, 100), (0x8030, 100), (0xC020, 100), (0xE020, 65535), (0x5040, 100), (0x5020, 100), (0xD010, 100), (0x7010, 4)]
>   string output type:  MT: Write message id 0x8E (SetStringOutputType) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x8F (SetStringOutputTypeAck) with 2 data bytes: [00 00]
> 0
>   period:  MT: Write message id 0x04 (SetPeriod) with 0 data bytes: []
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0x05 (SetPeriodAck) with 2 data bytes: [04 80]
> 1152
>   alignment rotation sensor:  MT: Write message id 0xEC (SetAlignmentRotation) with 1 data bytes: [00]
> waiting for 1 bytes, got 0 so far: []
> MT: Got message id 0xED (SetAlignmentRotationAck) with 16 data bytes: [3F 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
> Traceback (most recent call last):
>   File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1902, in <module>
>     main()
>   File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1547, in main
[... cut ...] rate)
>   File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1666, in inspect
>     parameter=0)
>   File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1642, in try_message
>     pprint.pprint(f(*args, **kwargs), indent=4)
>   File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 428, in GetAlignmentRotation
>     param, q0, q1, q2, q3 = struct.unpack('!Bffff', data)
> struct.error: unpack requires a string argument of length 17
>

fcolas, 2018-06-14:
@willdzeng Ok I see, thanks. They changed (fixed) the behaviour in `GetAlignmentRotation` after your version. I'll come up with a patch.

ozakiryota, 2018-06-19:
I still have the same issue I cannot get neither angular_velocity nor linear_acceleration like this,

`roslaunch xsens_driver xsens_driver.launch`
`rostopic echo /imu/data`
> header: 
  seq: 312
  stamp: 
    secs: 1529377005
    nsecs: 255074024
  frame_id: /imu
orientation: 
  x: -0.00360706239007
  y: -0.00864999182522
  z: -0.641639351845
  w: 0.766949117184
orientation_covariance: [0.01745, 0.0, 0.0, 0.0, 0.01745, 0.0, 0.0, 0.0, 0.15708]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]

I'm using MTi-30

Thanks

fcolas, 2018-06-20:
@0xd1ma This issue is about missing the linear velocity from GPS on MTi-G-710. Ensure in the configuration of your IMU that angular velocity and linear acceleration are supposed to be output (see my first comment above). If that's the case and you don't get any value, please open a new issue.

ozakiryota, 2018-06-20:
@fcolas<br>
I am sorry that I misunderstood and missed some.<br>
Thank you for the prompt and polite reply!

fcolas, 2018-06-21:
@willdzeng I push a new branch `fix_get_alignment_rotation`. Could you try it out to see whether it fixes the inspect issue?

willdzeng, 2018-06-26:
I tried it with your branch, `fix_get_alignment_rotation`, but it's getting same error:

> MT: Got message id 0xED (SetAlignmentRotationAck) with 16 data bytes: [3F 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
Traceback (most recent call last):
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1890, in <module>
    main()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1535, in main
    inspect(mt, device, baudrate)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1654, in inspect
    parameter=0)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1630, in try_message
    pprint.pprint(f(*args, **kwargs), indent=4)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 425, in GetAlignmentRotation
    param, q0, q1, q2, q3 = struct.unpack('!Bffff', data)
struct.error: unpack requires a string argument of length 17

I think that commit has already checked into the master, did you forget to push a new update on that branch?

fcolas, 2018-06-26:
Sorry, I messed up with my branches: now it should work better. Can you try it again?

willdzeng, 2018-06-27:
yup, that works.

fcolas, 2018-07-16:
@nflsalex Did you succeed?

nflsalex, 2018-07-31:
yes thanks!

closed 2018-07-31 by nflsalex