ethz-asl/ethzasl_xsens_driver #74: fixed point precision not supported.

willdzeng, 2018-06-08:
I got the following error when running the driver from roslaunch
```
[INFO] [1528494566.684170]: Found parameter: ~device, value: /dev/ttyUSB0
[INFO] [1528494566.687142]: Found parameter: ~baudrate, value: 0
[INFO] [1528494566.690184]: Found parameter: ~timeout, value: 0.002
[INFO] [1528494566.713816]: MT node interface: /dev/ttyUSB0 at 115200 bd.
[INFO] [1528494566.794297]: Found parameter: ~no_rotation_duration, value: 0
[INFO] [1528494566.797536]: Found parameter: ~frame_id, value: /imu
[INFO] [1528494566.801692]: Found parameter: ~frame_local, value: ENU
Traceback (most recent call last):
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 726, in <module>
    main()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 722, in main
    driver.spin()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 131, in spin
    self.spin_once()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 638, in spin_once
    data = self.mt.read_measurement()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 632, in read_measurement
    return self.parse_MTData2(data)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 912, in parse_MTData2
    raise MTException("fixed point precision not supported.")
mtdef.MTException: fixed point precision not supported.
```

This is what I got from mtdevice.py -i
```
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x077010E6
  product code: 'MTi-G-710-2A5G4'
  firmware revision: (1, 4, 12)
  baudrate: 2
  error mode:  message unsupported by your device.
  option flags:  0x00000000
  location ID:  0x0000
  transmit delay:  message unsupported by your device.
  synchronization settings:  [(0x09, 0x01, 0x01, 0x00, 0x0000, 0x0000, 0x0000, 0x00FA)]
  general configuration: {   'Master device ID': 124784870,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 124784870,
    'length': 0,
    'number of devices': 1,
    'output-mode': 6,
    'output-settings': 1,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  [(0x1020, 65535), (0x1060, 65535), (0x1010, 65535), (0x2030, 100), (0x4020, 100), (0x4010, 100), (0x8020, 100), (0x8030, 100), (0xE020, 65535), (0x5042, 100), (0x5022, 100), (0xD012, 100), (0x7010, 4), (0x8840, 4), (0x1030, 65535)]
  string output type: 0
  period: 1152
  alignment rotation sensor: 
Traceback (most recent call last):
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1903, in <module>
    main()
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1548, in main
    inspect(mt, device, baudrate)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1667, in inspect
    parameter=0)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 1643, in try_message
    pprint.pprint(f(*args, **kwargs), indent=4)
  File "/home/zeng/ros_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 428, in GetAlignmentRotation
    param, q0, q1, q2, q3 = struct.unpack('!Bffff', data)
struct.error: unpack requires a string argument of length 17
```

This is what I got from mtdevice.py
```
{'Acceleration': {'Delta v.z': 0.09774729609489441, 'Delta v.y': 0.0019016989972442389, 'Delta v.x': -0.005384126212447882, 'frame': 'ENU', 'accY': 0.19037173688411713, 'accX': -0.5384865403175354, 'accZ': 9.774721145629883}, 'Timestamp': {'Hour': 21, 'PacketCounter': 42124, 'Month': 6, 'Second': 17, 'Flags': 247, 'Year': 2018, 'ns': 734600000, 'Day': 8, 'Minute': 52, 'SampleTimeFine': 26266257}, 'Orientation Data': {'Yaw': 0.03150816261768341, 'Pitch': 3.0332014560699463, 'frame': 'ENU', 'Roll': 1.0015487670898438}, 'Angular Velocity': {'Delta q1': 1.0609558557916898e-05, 'Delta q0': 1.0, 'Delta q3': -5.172616511117667e-06, 'Delta q2': 3.682477199618006e-06, 'frame': 'ENU', 'gyrZ': -0.0010345234768465161, 'gyrX': 0.002121912082657218, 'gyrY': 0.0007364954799413681}, 'Status': {'StatusWord': 25166021}}

```

willdzeng, 2018-06-08:
Note that if I change https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtdevice.py#L916
to
```python
if (data_id & 0x0003) == 0x3:
  float_format = 'd'
elif (data_id & 0x0003) == 0x0:
  float_format = 'f'
else:
  break;
```
it works just fine

fcolas, 2018-06-11:
Indeed fixed point precision is not yet supported: a pull request to add support is welcome.
In the meantime, unless you have a good reason for it (I'd be curious to know), I'd suggest simply configuring your device with floating point numbers.

However, removing the `raise` statement as you did is **not** working just fine: the values you will get will be wrong.

willdzeng, 2018-06-11:
I used the GUI to configure the device to send floating point, now it works. Maybe we can have the driver to set up the device using floating point since we are not supporting it anyway?

willdzeng, 2018-06-13:
I see, thanks.

closed 2018-06-13 by willdzeng