ethz-asl/ethzasl_xsens_driver #75: SyncOut indication

koko19, 2018-06-21:
I have configured MTI 10 to send sync out signal, and checked using oscilloscope that it indeed does it correctly. However I am unable to find a way to detect in which message sent by the driver to ros did the sync occur. Is this possible and if yes, could you please explain how?

fcolas, 2018-06-21:
Disclaimer: I never used the synchronization settings and I'm not sure I find the documentation so clear. Also, I'm not sure what you exactly want to do, nor what is your configuration (you can help on this).
But from what I gathered, if properly configured (`TriggerIndication` function), bit 22 of the `StatusWord` should contain tell you when the triggered occurred. This `StatusWord` is not published as an independent topic (for lack of need and well-defined interface), but you can get it from the `imu_data_str` topic.
Can you check that it is the case?

koko19, 2018-06-26:
I have two cameras which I want to sync with IMU syncOut signal. Afterwards, in order to assign correct IMU measurements to images I have to know which IMU message contained sync signal. I have configured the IMU using the Xsens SDK to send syncOut every 50ms (20Hz) and I receive that signal on the oscilloscope, however the StatusWord in imu_data_str always remains the same. Here is an example:

data: "{'Acceleration': {'Delta v.z': -0.013541683554649353, 'Delta v.y': 0.020512893795967102,\
  \ 'Delta v.x': 0.00022651255130767822, 'frame': 'ENU', 'accY': 8.205488204956055,\
  \ 'accX': 0.08979009836912155, 'accZ': -5.4161858558654785}, 'Timestamp': {'PacketCounter':\
  \ 17525, 'SampleTimeFine': 5375072}, 'Angular Velocity': {'Delta q1': -2.9886141419410706e-05,\
  \ 'Delta q0': 1.0000001192092896, 'Delta q3': -3.993697464466095e-05, 'Delta q2':\
  \ -1.4726073459314648e-05, 'frame': 'ENU', 'gyrZ': -0.03194957971572876, 'gyrX':\
  \ -0.023908914998173714, 'gyrY': -0.011780858039855957}, 'Temperature': {'Temp':\
  \ 31.0}, '**Status': {'StatusWord': 4194305}}**"

StatusWord is always 4194305.

fcolas, 2018-06-26:
This value has bit 22 on with means, according to the document “When SyncOut is active this bit will rise to 1”.
Can you show the output of:

    rosrun xsens_driver mtdevice.py -i

fcolas, 2018-07-16:
@koko19 Did you find a way to achieve what you wanted?