ethz-asl/ethzasl_xsens_driver #77: Xsens Driver Implementation in C++

sumitspattan, 2018-06-26:
Hello,

I wanted to implement xsens ros driver in C++. What is the basic idea to proceed and how to make it ?

fcolas, 2018-06-26:
The basic idea is to look at the low-level communication documentation (now available on Xsens website), correctly parse the messages from the IMU (and send correct messages), and then transform them in appropriate ROS messages.
It is relatively easy to do if you restrict yourself to only a few features and models. It's more complex if you want to support many different devices with various configurations (all?) and protocol versions.

When I did it, the existing drivers didn't support the configurations I needed so I had to develop a new node. But I wonder why you want to do that now? Do you have a specific issue about the present node? Do you observe limitations in resources (memory or time)?

sumitspattan, 2018-06-27:
I am working in a company and we have development platform. So we wanted to implement xsens driver in C++. 
There is no problem with the present node.

fcolas, 2018-06-28:
Ok, I'm not sure I understand the reasoning but I hope you'll be able to do it with the information I provided. In any case, I will close this issue (because that's not a real issue).
You can still contact me here if you have specific questions.

closed 2018-06-28 by fcolas

romainreignier, 2018-11-09:
@sumitspattan Did you managed to implement this driver in C++? If so, do you have plans to open-source it?