ethz-asl/ethzasl_xsens_driver #83: Unable to change sampling frequency using MTdevice.py

salahkhan94, 2018-07-30:
I have a MTi-3-8A7G6 IMU, since the launch file throws  "could not find proper MT device" , typically I manually first rosrun mtdevice.py and then mtnode.py using the following bash scripts which just keep running commands every 1 sec until it succeds

```

#!/usr/bin/env bash
imu_config_cmd="rosrun xsens_driver mtdevice.py -i --configure="oq200,aa200,wr200" -b 921600"
#exec 1>/dev/null
#exec 2>&1
until ${imu_config_cmd}; do
  echo cmd failed, retrying in 1 second...
  sleep 1
done
echo "finished imu configuration"
```
```

#!/usr/bin/env bash
imu_run_cmd="rosrun xsens_driver mtnode.py _device:=/dev/ttyUSB0 _baudrate:=921600"
#exec 1>/dev/null
#exec 2>&1
until ${imu_run_cmd}; do
  echo cmd failed, retrying in 2 seconds...
  sleep 2
done
echo "finished imu run"
```

This works perfectly in publishing data except, I am unable to change the frequency /imu/data which is being published at 100 hz , My requirement is 200 Hz.

Following is the output of mtdevice.py -i
```

workstation@workstation:~/ros_client$ ./imu_config.sh
Device: /dev/ttyUSB0 at 921600 Bd:
  device ID:  0x03881CF7
  product code: 'MTi-3-8A7G6'
  firmware revision: (1, 2, 1)
  baudrate: 128
  error mode:  timeout: might be unsupported by your device.
  option flags:  0x00000000
  location ID:  0x0000
  transmit delay:  timeout: might be unsupported by your device.
  synchronization settings:  []
  general configuration: {   'Master device ID': 59251959,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 59251959,
    'length': 888,
    'number of devices': 1,
    'output-mode': 0,
    'output-settings': 0,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  [(0x2010, 100), (0x4020, 100), (0x8020, 100)]
  string output type:  timeout: might be unsupported by your device.
  period: 1152
  alignment rotation sensor: (0, (1.0, 0.0, 0.0, 0.0))
  alignment rotation local: (1, (1.0, 0.0, 0.0, 0.0))
  output mode:  0x0000
  extended output mode:  timeout: might be unsupported by your device.
  output settings:  0x00000000
  GPS coordinates (lat, lon, alt): (52.24049169408207, 6.838216771186439, 0.0)
  available scenarios: [   (50, 8, 'general'),
    (51, 8, 'high_mag_dep'),
    (52, 8, 'dynamic'),
    (53, 8, 'north_reference'),
    (54, 8, 'vru_general')]
  current scenario ID: 50
  UTC time: (0, 1970, 1, 1, 0, 0, 0, 0)
Changing output configuration  Ok
finished imu configuration

```

fcolas, 2018-08-01:
What is the command line you tried to change the output? Did you get an error?

salahkhan94, 2018-08-02:
I actually don't run the command line directly , I've put it inside a bash script and run that instead. Check the first bash script I wrote above.

Any who this is the command : rosrun xsens_driver mtdevice.py -i --configure="oq200,aa200,wr200" -b 921600"

fcolas, 2018-08-02:
Ok, there are several things about this first script.
First, it's advised not to write too often the onboard flash memory to reduce wear (see discussion in #70).
Second, you might want to check your quotes in that second line: I'd expect it tries to set the configuration empty.

salahkhan94, 2018-08-02:
Thanks, I guess I'll just run the mtdevice command separately and onl

fcolas, 2018-08-02:
> Could you add the command to this thread which could possibly change the frequency rate

Well, the command from your comment above should do the trick. Does it?
(Note that I'm not a support hotline; please read the documentation and try to solve your issue by yourself. I'm providing a node, I'm willing to improve it and to help using it but if you don't try and just ask me what you should do, I'll just close this thread as it doesn't seem to be an issue with this code.)

salahkhan94, 2018-08-02:
It doesn't do the trick , I get 100hz rate on imu/data . After I run the command I get "Changing output configuration  Ok" the baud rate sets at 921600 but frequency doesn't change to 200hz. I'll post here if I figure something out.

Apologies for the troubles. I'll close this.