ethz-asl/ethzasl_xsens_driver #84: support for mti-7?

aarontan94, 2018-07-30:
Hello, does this package work for [mti-7](https://shop.xsens.com/shop/offer/mti-7-development-kit-offer?utm_source=emark&utm_medium=email&utm_campaign=campaign-ism-q2-2-2018-newsletter)?

Thanks,
Aaron

nflsalex, 2018-07-31:
same question here

fcolas, 2018-08-01:
Well, I have no first-hand experience but, according to the documentation is still supports the same communication protocol (XBus) that the node is relying on.
They seem to support another protocol (MTSSP) on some interfaces. This one is not implemented by this package.

So I would advise you check in more details how you want to connect and use your MTi-7.

If you get to try one, I'd be happy to hear whether it works.

closed 2018-08-01 by fcolas

nflsalex, 2018-08-13:
I tried the package for the latest MTI-7-DK. It turns out I can only get the IMU data but not the GPS signal

fcolas, 2018-09-12:
@nflsalex I'm glad to hear it partly works.
Do you have an idea why you don't get GPS? Did you configure the device properly?
If you think it is a bug in the driver, please open a new issue specifically on this so we can investigate.