ethz-asl/ethzasl_xsens_driver #88: ResetOrientation(0x0001) -> Error 0x21 "An invalid parameter is supplied"

amastrobera, 2018-08-29:
I try to use the API to reset the heading, just like in the [GUI](https://base.xsens.com/hc/en-us/articles/360003435693-MT-Manager-Orientation-Reset-Options). I saw in the user manual that the request to launch should be done in either config or measurement mode, with `code = 0x0001`, and your API does this.
```
    def ResetOrientation(self, code):
        """Reset the orientation.
        Code can take several values:
            0x0000: store current settings (only in config mode),
            0x0001: heading reset (NOT supported by MTi-G),
            0x0003: object reset.
        """
        print "msg: 0x%02X, code: 0x%02X " % (MID.ResetOrientation, code)
        data = struct.pack('!H', code)
        self.write_ack(MID.ResetOrientation, data)
```

If I am not mistaken, there could be a problem with `packet += struct.pack('!B', 0xFF & (-(sum(map(ord, packet[1:])))))` in `mtdevice.py`. This request, as in DataView on the GUI ***should be*** `'\xfa\xff\xa4\x02\x00\x01\x5A'` but is actually `'\xfa\xff\xa4\x02\x00\x01Z'`. 

I think that is the beginning of an investigation.

amastrobera, 2018-08-29:
FYI @bdholt1 @LukePhairatt

amastrobera, 2018-09-12:
@fcolas has anyone looked at this ?

fcolas, 2018-09-12:
@amastrobera I have good confidence that the checksum code is correct here (because that's a direct transliteration of the manual and because otherwise there would have been issues since a while.
And in your particular examples, it happens that ascii for 'Z' is 5A thus both strings are identical.

For your actual issue, is it working with the GUI? Are you sure it's actually sending the request and not hiding error messages?

As a side note, please include the problem description in the body of the issue and not only in the title. That way, it is more legible and you can actually describe what is happening.

snakehaihai, 2019-01-12:
@amastrobera  Go windows download xsense suit. Open MT manager.  Increase baud rate to 921600. At output config, change to some default setting in the drop list. see if it works.  After that, select only the parameters you needed.