ethz-asl/ethzasl_xsens_driver #94: No linear velocity and angular data on the MTi device

aaravrav142, 2018-11-02:
Hi, 
I am using the newest version of MTi 30 AHRS bought just a few weeks ago. 

I can see the position data (x,y and z) but the node does not publish the velocity data (linear and angular). 

Below is the device information: 
```
$ rosrun xsens_driver mtdevice.py -i
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  0x0368248C
  product code: 'MTi-30-2A8G4'
  firmware revision: (1, 7, 2)
  baudrate: 2
  error mode:  message unsupported by your device.
  option flags:  0x00000000
  location ID:  0x0000
  transmit delay:  message unsupported by your device.
  synchronization settings:  []
  general configuration: {   'Master device ID': 57156748,
    'date': '\x00\x00\x00\x00\x00\x00\x00\x00',
    'device ID': 57156748,
    'length': 0,
    'number of devices': 1,
    'output-mode': 0,
    'output-settings': 0,
    'period': 1152,
    'skipfactor': 0,
    'time': '\x00\x00\x00\x00\x00\x00\x00\x00'}
  output configuration (mark IV devices):  [(0x1060, 65535), (0xE020, 65535), (0x2030, 100)]
  string output type: 0
  period: 1152
  alignment rotation sensor: (0, (1.0, 0.0, 0.0, 0.0))
  alignment rotation local: (1, (1.0, 0.0, 0.0, 0.0))
  output mode:  message unsupported by your device.
  extended output mode:  0x0000
  output settings:  message unsupported by your device.
  GPS coordinates (lat, lon, alt): (52.24049169408207, 6.838216771186439, 0.0)
  available scenarios: [   (39, 15, 'general'),
    (40, 15, 'high_mag_dep'),
    (41, 15, 'dynamic'),
    (42, 15, 'low_mag_dep'),
    (43, 15, 'vru_general')]
  current scenario ID: 39
  UTC time: (0, 1970, 1, 1, 0, 0, 0, 0)

```
and 
```
$ rosrun xsens_driver mtdevice.py -v -i
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x31 (GoToConfigAck) with 0 data bytes: []
MT: Write message id 0x0C (ReqConfiguration) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x0D (Configuration) with 118 data bytes: [03 68 24 8C 04 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 68 24 8C 00 00 00 00 00 00 00 00 00 27 01 07 02 49 05 01]
Device: /dev/ttyUSB0 at 115200 Bd:
  device ID:  MT: Write message id 0x00 (ReqDID) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x01 (DeviceID) with 4 data bytes: [03 68 24 8C]
0x0368248C
  product code:  MT: Write message id 0x1C (ReqProductCode) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x1D (ProductCode) with 20 data bytes: [4D 54 69 2D 33 30 2D 32 41 38 47 34 20 20 20 20 20 20 20 20]
'MTi-30-2A8G4'
  firmware revision:  MT: Write message id 0x12 (ReqFWRev) with 0 data bytes: []
waiting for 1 bytes, got 0 so far: []
MT: Got message id 0x13 (FirmwareRev) with 11 data bytes: [01 07 02 00 00 00 2A 00 00 FD 87]
(1, 7, 2)
  baudrate:  MT: Write message id 0x18 (SetBaudrate) with 0 data bytes: []
[... 255 lines cut ...]
  stamp: 
    secs: 1541144646
    nsecs: 336143016
  frame_id: "/imu"
orientation: 
  x: -0.000673421295964
  y: -0.00672165573111
  z: 0.100306365918
  w: 0.99493366553
orientation_covariance: [0.01745, 0.0, 0.0, 0.0, 0.01745, 0.0, 0.0, 0.0, 0.15708]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]
---

```

How do I solve this? Thanks

snakehaihai, 2019-01-12:
go to windows. Download xsense MT suit.  
$ At sensor config tab increase baudrate to 921600. 
$ At output config tab. enable delta t and delta v acceration. 

REMEMBER to APPLY and SAVE to MT before close

aaravrav142, 2019-01-15:
I do not have access to the sensor right now. I will report back soon. Thanks for the info.

snakehaihai, 2019-01-15:
OK do remember to uncheck other output such as 2000hz raw output and compass.    only check orientation, delta q and acceleration which the one you needed at 400hz

fcolas, 2019-08-25:
Note that you don't have to got to Windows and use the XSens configuration suite since you can use `mtdevice.py` to configure your IMU from the command line.

closed 2019-08-25 by fcolas