ethz-asl/ethzasl_xsens_driver #98: Populate position accuracy and horizontal accuracy as covariance

willdzeng, 2019-03-16:
Hi, 
Note that right now the driver publishes NavSatFix message but the position covariance are zeors.
I wonder if we can populate the position accuracy and horizontal accuracy as covariance?
I am happy to do some but can you provide some info on how would I do that?
Thanks.

```
position_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
position_covariance_type: 0
```

moooeeeep, 2019-05-28:
I've stumbled accross an approach to this here: https://github.com/westpoint-robotics/usma_xsens/blob/master/nodes/mtnode_new.py#L474

Besides that, it seems, you can set the other covariances via rosparam: https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtnode.py#L96

For a start, I'd suggest to add a parameter to specify the position covariance in the same way as the others.

fcolas, 2019-08-25:
I don't have time nor a device with a GPS to test, thus a PR is welcome (with either solution as long as it makes sense).