ethz-asl/ethzasl_xsens_driver #100: Driver Node publish topics at different Interval

zhouzhiqian, 2019-05-29:
When I use MTI300 on Ubunty 16.04 + ROS kinetic， the intervals between every two topics don't keep 5ms (its frequency is 200 Hz).   
Just like this figure:
[image]

But when I use it on Ubunty 14.04 + ROS indigo, it shows a totally different result.
[image]

)

fcolas, 2019-08-25:
That's a new issue for me.
Does it happen consistently or just sometimes (and if so, in which condition)?
I assume you checked it was not a CPU performance issue and only run the mtnode to reduce outside influences.
Is it the same system with two different installs or are there differences in the CPU/USB bus?

MichalNowicki, 2019-09-17:
I experienced this issue on a similar configuration and in our case, it was caused by the latency settings of the Ubuntu driver for FTDI. The solution was to set the low_latency mode of the driver and thus prevent it from accumulating multiple measurements.

The command to execute after the device was plugged in:
``setserial /dev/<tty_name> low_latency``

The setting resets once the device is disconnected but udev rules can be used to make it permanent.

zhouzhiqian, 2019-09-17:
@MichalNowicki  Thank you very much. I will try your solution quickly.

artemiialessandrini, 2019-10-08:
Hi, I am having the same issue of publisjing with different interval partially:

[image](https://yadi.sk/i/3K-aLn2AYzHXow)

While trying to  `setserial /dev/ttyUSB0 low_latency`, I get:
`Cannot get serial info: Inappropriate ioctl for device`

artemiialessandrini, 2019-10-08:
@zhouzhiqian was it a solution for you?

artemiialessandrini, 2019-10-08:
Apparently, rosbag_validate displays the same ERROR:
` E1008 14:32:31.060365 31665 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.660078 s, recommended is [0.0005, 0.005] s with no jitter.`