ethz-asl/ethzasl_xsens_driver #103: spin_once() redefines all the functions required for populating the output messages at every loop iteration

moooeeeep, 2019-08-23:
I just noticed that the function [`XSensDriver.spin_once`](https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtnode.py#L175), that does the parsing, populating and publishing at a potentially high frequency, redefines all the required helper functions locally at every loop iteration.

I would expect to receive a (significant?) speed-up, if these functions were converted to ordinary member functions or free functions and thus were only defined once for the runtime of the node.

It should be possible to rewrite the handler dispatching code [here](https://github.com/ethz-asl/ethzasl_xsens_driver/blob/master/nodes/mtnode.py#L717) in a straight-forward way using [`getattr()`](https://docs.python.org/3/library/functions.html#getattr). I can try to provide a PR, if I find the time.

fcolas, 2019-08-25:
You're right. I'm not sure how large the speedup can be but it is inefficient.
I would appreciate a PR when you have the time.

moooeeeep, 2019-09-16:
As stated in the comments in the [PR](https://github.com/ethz-asl/ethzasl_xsens_driver/pull/105) the inefficiency actually is insignificant compared to other factors. Therefore I'll close this issue.

closed 2019-09-16 by moooeeeep