ethz-asl/ethzasl_xsens_driver #107: Difference to or benefits of this project vs. official XSens MTi ROS driver?

moooeeeep, 2019-09-16:
I noticed, that XSens has released and does maintain an official ROS driver, downloadable as part of their SDK: http://wiki.ros.org/xsens_mti_driver

Did you already review the functionality of the official driver? Does it lack or differ in scope or features? Would you recommend to switch?

fcolas, 2019-09-16:
It's not a bad question and the short answer is that my package predates the official one (by many years).

I started this node because there was no official ROS (or even unix) driver and the third party driver was limited in functionalities (single configuration supported). Later XSens published a half-maintained ROS driver based on mine but they didn't commit to it. Hopefully they will do better with this one but I didn't try it (I don't use a XSens device any more).

I cannot comment on the details but the node seems to provide similar (and hopefully even better) functionality. One small difference in scope is that you can configure the device with my package; which was convenient when MT manager was only available on Windows. Now it's not the case any more but I believe a command line configuration tool can still be useful for robots.

Sadly, they don't seem to package it as part of the ROS distribution, which means more manual operations.

But if I don't have to do their job and they provide a stable, performant, full-featured and durable node, that's great.

I'd be interested in your opinion (or anybody else) on this question. In other words, should I deprecate this one and recommend to switch to the official package?

ruoshiwen, 2020-12-08:
I tried official package on ubuntu 16.04, but I could not run mtmanager. The error is libicui18n.so.60 cannot open shared object file, I spent quite some time searching the answer and still failed. 
Fortunately, your package works fine for me. Except that I had to modify the package.xml file, add <package format="2"> and changing <run_depend> to <exec_depend>, otherwise ros does not recognise the package name (cannot roslaunch), but I dont know why.

jiminghe, 2022-03-18:
Hi Everyone,

Xsens now publishes the latest version of Xsens ROS MTi Driver inside the MT Software Suite, please make sure that you download the latest MT Software Suite at: https://content.xsens.com/mt-software-suite-download