ethz-asl/ethzasl_xsens_driver #108: Unable to open /dev/*

samxchandlerpi, 2019-09-18:
I'm trying to get this driver working on a Jetson Nano with ROS melodic, but I can't seem to open the device. I've tried to play with the permissions and udev rules, but nothing seems to work.

**Error:**
> ... logging to /home/k10mini/.ros/log/d9bc322c-da40-11e9-8a43-f64025a2631d/roslaunch-k10mini-8249.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://192.168.55.1:44079/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: melodic
>  * /rosversion: 1.14.3
>  * /xsens_driver/device: /dev/IMU
> 
> NODES
>   /
>     xsens_driver (xsens_driver/mtnode.py)
> 
> auto-starting new master
> process[master]: started with pid [8262]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to d9bc322c-da40-11e9-8a43-f64025a2631d
> process[rosout-1]: started with pid [8274]
> started core service [/rosout]
> process[xsens_driver-2]: started with pid [8277]
> [INFO] [1568830773.347442]: Found parameter: ~device, value: /dev/IMU
> [WARN] [1568830773.353000]: Cannot find value for parameter: ~baudrate, assigning default: 0
> [WARN] [1568830773.357709]: Cannot find value for parameter: ~timeout, assigning default: 0.002
> Traceback (most recent call last):
>   File "/home/k10mini/ros/irlw-drivers/ethzasl_xsens_driver/nodes/mtnode.py", line 726, in <module>
>     main()
>   File "/home/k10mini/ros/irlw-drivers/ethzasl_xsens_driver/nodes/mtnode.py", line 721, in main
>     driver = XSensDriver()
>   File "/home/k10mini/ros/irlw-drivers/ethzasl_xsens_driver/nodes/mtnode.py", line 52, in __init__
>     baudrate = mtdevice.find_baudrate(device)
>   File "/home/k10mini/ros/irlw-drivers/ethzasl_xsens_driver/nodes/mtdevice.py", line 1133, in find_baudrate
>     raise MTException("unable to open %s" % port)
> mtdef.MTException: unable to open /dev/IMU
> [xsens_driver-2] process has died [pid 8277, exit code 1, cmd /home/k10mini/ros/irlw-drivers/ethzasl_xsens_driver/nodes/mtnode.py __name:=xsens_driver __log:=/home/k10mini/.ros/log/d9bc322c-da40-11e9-8a43-f64025a2631d/xsens_driver-2.log].
> log file: /home/k10mini/.ros/log/d9bc322c-da40-11e9-8a43-f64025a2631d/xsens_driver-2*.log
> ^C[rosout-1] killing on exit
> [master] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

**Permissions:**
> lrwxrwxrwx  1 root    root          15 Sep 18 14:14 IMU -> bus/usb/001/004
> crw-rw-rw- 1 root plugdev 189, 3 Sep 18 14:14 /dev/bus/usb/001/004

**udev rules:**
> ATTRS{idVendor}=="2639", ATTRS{idProduct}=="0002", ATTRS{serial}=="026001B1", MODE="0666", GROUP="plugdev", SYMLINK+="IMU"

fcolas, 2019-09-19:
The `serial` module cannot seem to be able to open your device.
You can try to set the `baudrate` to the value of your device and see if you have a different message. Or you can directly print the `serial.SerialException` to see what happens.

samxchandlerpi, 2019-09-19:
> The `serial` module cannot seem to be able to open your device.
> You can try to set the `baudrate` to the value of your device and see if you have a different message. Or you can directly print the `serial.SerialException` to see what happens.

I printed the serial.Serial Exception and received the following:

> Error: Could not configure port: (25, 'Inappropriate ioctl for device')

I did a little more digging by testing the output of dmesg for the computer I have the xsens device working on and this new computer that's returning the serial error

Here's the output of dmesg for the Jetson Nano which is **not** working:

> [11562.545478] usb 1-2.1: new full-speed USB device number 12 using tegra-xusb
> [11562.567294] usb 1-2.1: New USB device found, idVendor=2639, idProduct=0002
> [11562.567302] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [11562.567307] usb 1-2.1: Product: MTi-20 VRU
> [11562.567312] usb 1-2.1: Manufacturer: Xsens
> [11562.567317] usb 1-2.1: SerialNumber: 026001B1

Here's the output of dmesg for the computer that **is** working:

> [25131.692211] usb 1-2: new full-speed USB device number 15 using xhci_hcd
> [25131.841470] usb 1-2: New USB device found, idVendor=2639, idProduct=0002
> [25131.841475] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [25131.841478] usb 1-2: Product: MTi-20 VRU
> [25131.841481] usb 1-2: Manufacturer: Xsens
> [25131.841484] usb 1-2: SerialNumber: 026001B1
> [25131.842719] xsens_mt 1-2:1.1: xsens_mt converter detected
> [25131.843138] usb 1-2: xsens_mt converter now attached to ttyUSB0

So it seems that the Jetson Nano is not able to attach the xsens device to the ttyUSB* port. My guess is that this is probably due to a missing software driver for the xsens USB-Serial cable.

I tried to install this driver from here https://github.com/xsens/xsens_mt, following these instructions https://base.xsens.com/hc/en-us/articles/202302142-Restoring-communication-with-your-MTi but I had no luck.

I know the xsens software suite doesn't officially support arm64 devices, but I thought I could still get this working with your software since it doesn't depend on MT SDK. Looks like that may not be the case.

samxchandlerpi, 2019-09-20:
Okay, I finally got this working. I wasn't installing  https://github.com/xsens/xsens_mt drivers correctly.

Here's what I did:

1. Cloned https://github.com/xsens/xsens_mt into /lib/modules/kernel*/source/drivers/usb
2. cd xsens_mt
3. sudo make
4. modified first line of xsens_mt/modules.order to kernel/kernel*/drivers/usb/xsens_mt/xsens_mt.ko
5. cd /lib/modules/kernel*/kernel/drivers/usb
6. sudo mkdir xsens_mt
7. sudo cp /lib/modules/kernel*/source/drivers/usb/xsens_mt /xsens_mt.ko ./
8. sudo depmod -a
9. sudo vim /etc/modules
10. Add xsens_mt to end of the file
11. sudo insmod /lib/modules/kernel*/source/drivers/usb/xsens_mt.ko
12. restart

I doubt this is the easiest way to do it, but this is what worked from me. Some steps may also be redundant

FYI this is working on the Jetson Nano which is an ARM64 architecture. I'm don't think the official xsens software works because it requires the SDK which doesn't run on ARM. Thank you for developing this!

closed 2019-09-20 by samxchandlerpi

fcolas, 2019-09-20:
You're welcome, thanks for sharing your solution.