ethz-asl/ethzasl_xsens_driver #116: MTI-630 output timestamp interval trouble

guo-zixuan, 2021-03-29:
Hello guys!
My team focus on multi-sensor fusion slam.We found a confusing  phenomenon that the output of xsense driver is not evenly distributed in a time duration.For example,velodyne data is 10hz,xsense data is 100hz but xsense data get together in a time duration.
So, can I do some change in .launch file to let it evenly distribute in a time duration?
[image]

lipascal123, 2021-06-01:
Hi there,

I have the exact same issue. Did you find any solution? 
I'm also using it for multi-sensor fusion.

Cracks-ll, 2021-12-09:
@guo-zixuan 老哥你解决这个问题了吗？我用的是官方驱动，但是感觉官方驱动是错的