ethz-asl/ethzasl_xsens_driver #120: It doesn't work on Ubuntu 16.04

bigbigpark, 2022-08-30:
I'm using `Xsens MTi-30 AHRS`. <br/>
I downloaded xsens driver from [ethzasl_xsens_driverhttps://github.com/ethz-asl/ethzasl_xsens_driver) <br/>
when I type the command,
~~~bash
$ roslaunch xsens_driver xsens_driver.launch
~~~
An error occured, the log messages are below:
~~~bash
$ roslaunch xsens_driver xsens_driver.launch 
... logging to /home/mi/.ros/log/e9036b12-2833-11ed-a900-003064588f58/roslaunch-mi-9980.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:35433/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /xsens_driver/angular_velocity_covariance_diagonal: [0.0004, 0.0004, ...
 * /xsens_driver/baudrate: 0
 * /xsens_driver/device: auto
 * /xsens_driver/frame_id: /imu
 * /xsens_driver/frame_local: ENU
 * /xsens_driver/initial_wait: 0.1
 * /xsens_driver/linear_acceleration_covariance_diagonal: [0.0004, 0.0004, ...
 * /xsens_driver/no_rotation_duration: 0
 * /xsens_driver/orientation_covariance_diagonal: [0.01745, 0.01745...
 * /xsens_driver/timeout: 0.002

NODES
  /
    xsens_driver (xsens_driver/mtnode.py)

auto-starting new master
process[master]: started with pid [9990]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e9036b12-2833-11ed-a900-003064588f58
WARNING: Package name "RTK" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosout-1]: started with pid [10003]
started core service [/rosout]
process[xsens_driver-2]: started with pid [10012]
[INFO] [1661843894.696038]: Found parameter: ~device, value: auto
[INFO] [1661843894.696825]: Found parameter: ~baudrate, value: 0
[INFO] [1661843894.697652]: Found parameter: ~timeout, value: 0.002
[INFO] [1661843894.698488]: Found parameter: ~initial_wait, value: 0.1
[INFO] [1661843897.520299]: Detected MT device on port /dev/ttyUSB1 @ 115200 bps
[INFO] [1661843897.521144]: MT node interface: /dev/ttyUSB1 at 115200 bd.
[INFO] [1661843897.833314]: Found parameter: ~no_rotation_duration, value: 0
[INFO] [1661843897.836124]: Found parameter: ~frame_id, value: /imu
[INFO] [1661843897.838947]: Found parameter: ~frame_local, value: ENU
[INFO] [1661843897.841714]: Found parameter: ~angular_velocity_covariance_diagonal, value: [0.0004, 0.0004, 0.0004]
[INFO] [1661843897.844540]: Found parameter: ~linear_acceleration_covariance_diagonal, value: [0.0004, 0.0004, 0.0004]
[INFO] [1661843897.847618]: Found parameter: ~orientation_covariance_diagonal, value: [0.01745, 0.01745, 0.15708]
Traceback (most recent call last):
  File "/home/mi/scpark_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 803, in <module>
    main()
  File "/home/mi/scpark_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 799, in main
    driver.spin()
  File "/home/mi/scpark_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 168, in spin
    self.spin_once()
  File "/home/mi/scpark_ws/src/ethzasl_xsens_driver/nodes/mtnode.py", line 710, in spin_once
    data = self.mt.read_measurement()
  File "/home/mi/scpark_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 696, in read_measurement
    mid, data = self.read_msg()
  File "/home/mi/scpark_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 163, in read_msg
    raise MTException("could not find message.")
mtdef.MTException: could not find message.
[xsens_driver-2] process has died [pid 10012, exit code 1, cmd /home/mi/scpark_ws/src/ethzasl_xsens_driver/nodes/mtnode.py __name:=xsens_driver __log:=/home/mi/.ros/log/e9036b12-2833-11ed-a900-003064588f58/xsens_driver-2.log].
log file: /home/mi/.ros/log/e9036b12-2833-11ed-a900-003064588f58/xsens_driver-2*.log
~~~
What should I do? <br/>

Best regards.

peci1, 2023-04-19:
Try to set the driver to verbose mode (you can force it for example on this line: https://github.com/ethz-asl/ethzasl_xsens_driver/blob/f9bf1c9f02c266c04880d01a8e8be949507e94f3/nodes/mtdevice.py#L26 ). Post the output.

DrZ-21, 2023-07-18:
@bigbigpark Hi, have you solved this problem? I also met this question. How did you do？

bigbigpark, 2023-07-18:
Pretty far,  I cannot remember what I did.

I could not find any solutions on Ubuntu 16.04. So I upgraded my system on Ubuntu 18.04

peci1, 2023-07-25:
> @bigbigpark Hi, have you solved this problem? I also met this question. How did you do？

https://github.com/ethz-asl/ethzasl_xsens_driver/issues/120#issuecomment-1514836910